open-mmlab/OpenPCDet #1356: Issues with empty label files

GaneshAdam, 2023-05-22:
Hi Author,

Thank you for this repo!
I have dataset with frames having empty labels in both test and train set and I want to include them.
At present frames with empty labels are excluded by doing resampling so frame with empty labels is not considered for training?. same can be seen in file "pcdet/datasets/dataset.py" as
if self.training and len(data_dict['gt_boxes']) == 0:
            new_index = np.random.randint(self.__len__())
            return self.__getitem__(new_index)

 I feel, including frames with empty labels can help balance the dataset, can influence the model's behavior and help it distinguish between objects and non-objects.

closed 2023-07-06 by github-actions[bot]

tyutlk, 2023-10-10:
Hi,have u solved it?

Petros626, 2025-01-14:
@GaneshAdam @tyutlk if you take a look into the label files (.txt) you can see that objects are annotated. I don't understand why in `kitti_infos_train.pkl` the lidar_idx xy contains values in `gt_boxes_lidar` and later not.

GaneshAdam, 2025-01-15:
@Petros626 They are later removed based on pointcloud range you choose for training. For e.g. if you are training with +-80m in both x and y, then if any GT outside this range will be filtered. So, if a particular frame has all objects in its label file are outside this range then ideally they will be filtered and this frame will have empty GTs.

GaneshAdam, 2025-01-15:
> Hi,have u solved it?

Hi,
Basically if GT labels for particular frame are empty or not is not checked in code,  I solved it by adding/updating code(majorly checking if GT objects are empty and handle gt objects processing code accordingly) in files like 
     dataset.py 
     augmentor_utils.py
     data_agmentor.py
     eval.py 
     kitti_dataset.py(I majorly use kitti format, you might need to change in other dataset py files too if needed)
     data_processor.py
     anchor_head_single.py
      common_utils.py

Petros626, 2025-01-15:
> @Petros626 They are later removed based on pointcloud range you choose for training. For e.g. if you are training with +-80m in both x and y, then if any GT outside this range will be filtered. So, [...]

Yes exactly I later understood the whole process so this is correct.

Petros626, 2025-01-16:
@GaneshAdam I came across the question if the new added index is in the range of 3712 samples from train.txt, then so we got doubled samples in our training data?

GaneshAdam, 2025-01-18:
> [@GaneshAdam](https://github.com/GaneshAdam) I came across the question if the new added index is in the range of 3712 samples from train.txt, then so we got doubled samples in our training data?

I didn't really understand your query

Petros626, 2025-01-18:
If the sample isn't in the defined range of `POINT_CLOUD_RANGE: [0, -30, -2.73, 60, 30, 1.27]` the sample_id gets replaced by a random new one from the `train.txt` file. This can occur by chance doubled sample_id's:

```txt
Original: 0000014 -> replaced 003814
Original: 0004151 -> replaced 003814
```

GaneshAdam, 2025-01-18:
> If the sample isn't in the defined range of `POINT_CLOUD_RANGE: [0, -30, -2.73, 60, 30, 1.27]` the sample_id gets replaced by a random new one from the `train.txt` file. This can occur by chance doubled sample_id's:
> 
> Original: 0000014 -> replaced 003814
> Original: 0004151 -> replaced 003814

Yes This is possible.