open-mmlab/OpenPCDet #1363: waymo v1.0，AP result  all 0

carrolltm, 2023-05-26:
Hi, I  have this problem when I use waymo v1.0 for object detection(Start the waymo evaluation... Number:(pd, 1322657)VS. (gt,0))，i also use the environment，Can you tell me how you finally solved it?Thank u very much!!!
image
[image]
[image]

jihanyang, 2023-05-31:
It is very strange. I haven't tried Waymo 1.0 for a long time, but it should work nomally. Can you provide more information such as pcdet version, environment, etc.

closed 2023-07-16 by github-actions[bot]

sinatayebati, 2024-02-06:
@carrolltm I have the same issue! Have you solved the problem?

carrolltm, 2024-02-06:
please use the waymov1.2

---- Replied Message ----
| From | Sina ***@***.***> |
| Date | 02/07/2024 07:33 |
| To | open-mmlab/OpenPCDet ***@***.***> |
| Cc | carroll ***@***.***>,
Mention ***@***.***> |
| Subject | Re: [open-mmlab/OpenPCDet] waymo v1.0，AP result all 0 (Issue #1363) |

@carrolltm I have the same issue! Have you solved the problem?

sinatayebati, 2024-02-08:
> please use the waymov1.2
> […](#)

great thanks! I realized in v1.2 validation data there is a folder called "ground_truth_objects" which I assume is what's needed. Could you please guide me where I should put this in the data structure before producing the data infos? 

There's nothin mentioned about this particular file and where it should be in [openpcdet waymo dataset](https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md#waymo-open-dataset).