open-mmlab/OpenPCDet #1365: HELP! low accuracy on once dataset using pointrcnn.

DurbinLiu, 2023-05-30:
I trained pointrcnn on once dataset for 80 epochs with the default config.
But got very poor accuracy as follows:
[image]
I wonder where's the problem, did I set the config wrong?
could u please give me some advice on how to improve the performance?
Many thanks!

PointsCoder, 2023-06-09:
The scores are as expected and similar to what was reported in the [paper](https://arxiv.org/pdf/2106.11037.pdf).

To further improve, you may try changing to a smaller batch size or adjusting the ball query radius, which might help a little.

DurbinLiu, 2023-06-12:
> 

Hi, thanks for your reply, why smaller batch size would help? do you mean set a bigger batch size?

PointsCoder, 2023-06-30:
I didn't try but I heard using batch size 1 can boost the AP by 2%
https://github.com/PointsCoder/ONCE_Benchmark/issues/31#issuecomment-1311804549

PointsCoder, 2023-06-30:
I guess it is because the training set is small so different batch sizes bring some randomness on the performances.

closed 2023-08-15 by github-actions[bot]