open-mmlab/OpenPCDet #1366: shuffle=False in Dataloader does not work

zhangtingyu11, 2023-05-30:
I want to train the model as the original orders, and I change [shuffle=(sampler is None) and training](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/__init__.py#79) to `shuffle=False`, but it does not work, the frame id is also random. Is there any reason behind this? Hoping for your reply.

zhangtingyu11, 2023-05-31:
> Would you like to provide more details, like the order, dataset, and order before/after such change?

When I set shuffle =False, it seems that the first sample idx is random, and the following sample idx is in sequence.

jihanyang, 2023-06-01:
That's pretty strange for me. I haven't met this before.

zhangtingyu11, 2023-06-02:
> That's pretty strange for me. I haven't met this before.

I found the reason in [this line](https://github.com/open-mmlab/OpenPCDet/blob/02ac3e17a9aa438bdab9b566d4334bff6263374a/pcdet/datasets/dataset.py#L212). The first index is randomly chosen because I only train the car class and there is no car in frame id 000000.

closed 2023-07-17 by github-actions[bot]