open-mmlab/OpenPCDet #1367: Question about Spconv1.2  （Segmentation fault (core dumped)）

CBY-9527, 2023-05-31:
My environment is ubuntu20.04, 3090, cuda11.3, python3.7, pytorch1.7. spconv1.2 and pcdet were installed successfully, but the segmentation fault was dumped at the time of training. Mispositioning should be a problem with spconv. Anyone who has encountered this problem or has a solution.

jihanyang, 2023-05-31:
I strongly recommand you to use spconv2.x in your case.

jihanyang, 2023-05-31:
You means the training performance or the evaluation performance?

CBY-9527, 2023-05-31:
Thank you for your reply. I trained ROS on the nuscenes dataset, and then directly tested KITTI to obtain the above performance.

jihanyang, 2023-06-01:
I mean, when you evaluate the official pretrained models, have you found the validation results difference between spconv1.2 and spconv2.x?

CBY-9527, 2023-06-06:
> I mean, when you evaluate the official pretrained models, have you found the validation results difference between spconv1.2 and spconv2.x?

Thanks for your reply, 3D sparse convolution parameters cannot be loaded directly (spconv1.2-->spconv 2.X).
Recently, I solved the problem of installing spconv1.2 on ubuntu20.04 with 3090GPU, and the model can now be trained normally.

closed 2023-07-22 by github-actions[bot]