open-mmlab/OpenPCDet #1374: Can not load evaluation dataset.

wangyakunn, 2023-06-08:
When I run train.py, a ckpt can be successfully trained but evaluation is failed. I find the problem in build_dataloader().
```
train_set, train_loader, train_sampler = build_dataloader(
        dataset_cfg=cfg.DATA_CONFIG,
        class_names=cfg.CLASS_NAMES,
        batch_size=args.batch_size,
        dist=dist_train, workers=args.workers,
        logger=logger,
        training=True,
        merge_all_iters_to_one_epoch=args.merge_all_iters_to_one_epoch,
        total_epochs=args.epochs
    )
```
```
test_set, test_loader, sampler = build_dataloader(
        dataset_cfg=cfg.DATA_CONFIG,
        class_names=cfg.CLASS_NAMES,
        batch_size=args.batch_size,
        dist=dist_train, 
        workers=args.workers, 
        logger=logger, 
        training=False
    )
```
The param **training** which is set to **True** can load data successfully but False cause a mistake.(len(train_loader))！=0，len(test_loader)=0).How fix it?

OrangeSodahub, 2023-06-10:
It means your test split pkl is empty.

wangyakunn, 2023-06-10:
> It means your test split pkl is empty.

closed 2023-06-10 by wangyakunn