open-mmlab/OpenPCDet #1375: bevfusion results

DezeZhao, 2023-06-09:
how many epochs have you trained on BEVFusion?  I trained 6 epochs but get lower performance than you. Thanks.

chenshi3, 2023-06-09:
6 epochs. Could you report your results?

DezeZhao, 2023-06-09:
2023-06-09 05:54:28,991   INFO  *************** EPOCH checkpoint_epoch_6 EVALUATION *****************
2023-06-09 05:58:03,049   INFO  *************** Performance of EPOCH checkpoint_epoch_6 *****************
2023-06-09 05:58:03,049   INFO  Generate label finished(sec_per_example: 0.0012 second).
2023-06-09 05:58:03,049   INFO  recall_roi_0.3: 0.000000
2023-06-09 05:58:03,049   INFO  recall_rcnn_0.3: 0.798415
2023-06-09 05:58:03,049   INFO  recall_roi_0.5: 0.000000
2023-06-09 05:58:03,049   INFO  recall_rcnn_0.5: 0.593576
2023-06-09 05:58:03,049   INFO  recall_roi_0.7: 0.000000
2023-06-09 05:58:03,049   INFO  recall_rcnn_0.7: 0.266604
2023-06-09 05:58:03,053   INFO  Average predicted number of objects(6019 samples): 200.000

2023-06-09 06:05:55,304   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.17, 0.15, 0.06, 0.31, 0.20 | 79.72, 89.10, 91.66, 92.73 | mean AP: 0.8830172234914054
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.37, 0.20, 0.10, 0.29, 0.23 | 35.18, 55.36, 66.79, 72.46 | mean AP: 0.5744712429517645
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.69, 0.46, 0.92, 0.14, 0.31 | 2.25, 22.12, 39.97, 47.51 | mean AP: 0.27962523349293067
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.37, 0.19, 0.09, 0.62, 0.25 | 43.27, 67.62, 84.73, 89.00 | mean AP: 0.7115396726673869
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.52, 0.23, 0.69, 0.29, 0.19 | 11.99, 35.72, 52.30, 60.57 | mean AP: 0.4014552906321322
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.19, 0.28, 0.06, nan, nan | 59.52, 70.15, 74.13, 75.40 | mean AP: 0.6980134584812474
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.21, 0.26, 0.47, 0.42, 0.23 | 59.89, 73.01, 75.44, 76.60 | mean AP: 0.7123434972417897
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.18, 0.26, 0.76, 0.21, 0.02 | 50.28, 55.88, 56.49, 58.03 | mean AP: 0.5516955496353284
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.14, 0.29, 0.43, 0.25, 0.11 | 84.87, 86.12, 87.17, 88.23 | mean AP: 0.8659750823258545
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.13, 0.34, nan, nan, nan | 74.63, 75.99, 78.16, 80.48 | mean AP: 0.7731555530648122
--------------average performance-------------
trans_err:	 0.2971
scale_err:	 0.2660
orient_err:	 0.3984
vel_err:	 0.3180
attr_err:	 0.1912
mAP:	 0.6451
NDS:	 0.6755

DezeZhao, 2023-06-09:
> 2023-06-09 05:54:28,991 INFO *************** EPOCH checkpoint_epoch_6 EVALUATION ***************** 2023-06-09 05:58:03,049 INFO *************** Performance of EPOCH checkpoint_epoch_6 [...]

lower 3%, with 6x 3090 gpus, batchsize = 3 per gpu

chenshi3, 2023-06-09:
The NDS you report is lower than that of Transfusion-L. I suggest retraining the model again as the gradient of Transfusion-Head is not stable, and I have encountered similar issue. Additionally, the ckpts provided are trained with spconv 1.0 and I observed performance drop when switching to spconv 2.0.

DezeZhao, 2023-06-09:
ok thanks for your clue and quick reply.

closed 2023-06-09 by DezeZhao