open-mmlab/OpenPCDet #1381: Regarding the reproduction of BEVFUSION, the final evaluation results are inconsistent with the official results

hehualin-tut, 2023-06-12:
Hello, I tried to reproduce bevfusion, I used the officially trained model, but in the end I found that the results did not correspond to the official。
I'm using pytorch1.9.0+cu10.2 spconv1.2
here is my log

[log_eval_20230612-140648.txt](https://github.com/open-mmlab/OpenPCDet/files/11718910/log_eval_20230612-140648.txt)
I intercepted the results as follows:
_--------------average performance-------------
trans_err:	 0.4859
scale_err:	 0.5822
orient_err:	 0.7831
vel_err:	 0.6876
attr_err:	 0.2429
mAP:	 0.1141
NDS:	 0.2789_

You can see that mAP is a lot different from NDS and official 67.75 and 79.98, I don't know where the problem is
The only thing I modified was change the`torch.floor_div(self,other)` to `torch.div(self,other,rounding_model='floor')`

chenshi3, 2023-06-12:
Base on your log, you use assigned `pretrained_model` as `/data/1hhl/cbgs_transfusion_lidar.pth`, which will reload the ckpt when evaluating.

hehualin-tut, 2023-06-12:
> Base on your log, you use assigned `pretrained_model` as `/data/1hhl/cbgs_transfusion_lidar.pth`, which will reload the ckpt when evaluating.

solved,thank you. i will close this issue

closed 2023-06-12 by hehualin-tut