open-mmlab/OpenPCDet #1388: In bev_pool(feats, coords, B, D, H, W) what B,D parameter stand for ?

ammaryasirnaich, 2023-06-18:
I have a 3D voxel feature that i want to convert to BEV I think the function that can help is the function `def bev_pool(feats, coords, B, D, H, W)` in the **pcdet/ops/bev_pool/bev_pool**. I do understand that H,W must stand for **hight** and **weight** but don't know what B, D stands for ? Would appreciate if anyone can guide me.

Thanks

chenshi3, 2023-06-19:
B and D stand for batch-size and Z-dimension respectively.

ammaryasirnaich, 2023-06-19:
@chenshi3, many thanks for the answer. But usually, what comes in `D` Z-dimension, after giving `feature` and `coordinates`?  Although the coordinates do have varying Z-dimensions.

chenshi3, 2023-06-19:
> @chenshi3, many thanks for the answer. But usually, what comes in `D` Z-dimension, after giving `feature` and `coordinates`? Although the coordinates do have varying Z-dimensions.

D is computed as (Z_Up_Bound-Z_Low_Bound)/Voxel_z.

ammaryasirnaich, 2023-06-20:
> > @chenshi3, many thanks for the answer. But usually, what comes in `D` Z-dimension, after giving `feature` and `coordinates`? Although the coordinates do have varying Z-dimensions.
> 
> D is computed as (Z_Up_Bound-Z_Low_Bound)/Voxel_z.

Many thanks, @chenshi3 for helping out, really appreciate it

closed 2023-06-20 by ammaryasirnaich