open-mmlab/OpenPCDet #1392: ValueError: need at least one array to stack

Choubey176, 2023-06-22:
I am trying to train the point-rcnn model with kitti dataset but getting this issue.... please help me to solve

2023-06-22 10:24:04,825   INFO  **********************Start logging**********************
2023-06-22 10:24:04,826   INFO  CUDA_VISIBLE_DEVICES=ALL
2023-06-22 10:24:04,826   INFO  Training with a single process
2023-06-22 10:24:04,826   INFO  cfg_file         /content/OpenPCDet/tools/cfgs/kitti_models/pointrcnn.yaml
2023-06-22 10:24:04,826   INFO  batch_size       2
2023-06-22 10:24:04,826   INFO  epochs           80
2023-06-22 10:24:04,826   INFO  workers          4
2023-06-22 10:24:04,826   INFO  extra_tag        default
2023-06-22 10:24:04,826   INFO  ckpt             None
2023-06-22 10:24:04,826   INFO  pretrained_model None
2023-06-22 10:24:04,826   INFO  launcher         none
2023-06-22 10:24:04,826   INFO  tcp_port         18888
2023-06-22 10:24:04,826   INFO  sync_bn          False
2023-06-22 10:24:04,826   INFO  fix_random_seed  False
2023-06-22 10:24:04,826   INFO  ckpt_save_interval 1
2023-06-22 10:24:04,826   INFO  local_rank       0
2023-06-22 10:24:04,826   INFO  max_ckpt_save_num 30
2023-06-22 10:24:04,827   INFO  merge_all_iters_to_one_epoch False
2023-06-22 10:24:04,827   INFO  set_cfgs         None
2023-06-22 10:24:04,827   INFO  max_waiting_mins 0
2023-06-22 10:24:04,827   INFO  start_epoch      0
2023-06-22 10:24:04,827   INFO  num_epochs_to_eval 0
2023-06-22 10:24:04,827   INFO  save_to_file     False
2023-06-22 10:24:04,827   INFO  use_tqdm_to_record False
2023-06-22 10:24:04,827   INFO  logger_iter_interval 50
2023-06-22 10:24:04,827   INFO  ckpt_save_time_interval 300
2023-06-22 10:24:04,827   INFO  wo_gpu_stat      False
2023-06-22 10:24:04,827   INFO  use_amp          False
2023-06-22 10:24:04,827   INFO  cfg.ROOT_DIR: /content/OpenPCDet
2023-06-22 10:24:04,827   INFO  cfg.LOCAL_RANK: 0
2023-06-22 10:24:04,827   INFO  cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']
2023-06-22 10:24:04,827   INFO  ----------- DATA_CONFIG -----------
2023-06-22 10:24:04,827   INFO  cfg.DATA_CONFIG.DATASET: KittiDataset
2023-06-22 10:24:04,827   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/kitti
2023-06-22 10:24:04,827   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1]
2023-06-22 10:24:04,828   INFO  ----------- DATA_SPLIT -----------
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2023-06-22 10:24:04,828   INFO  ----------- INFO_PATH -----------
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_val.pkl']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.FOV_POINTS_ONLY: True
2023-06-22 10:24:04,828   INFO  ----------- DATA_AUGMENTOR -----------
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': True, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2023-06-22 10:24:04,828   INFO  ----------- POINT_FEATURE_ENCODING -----------
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'sample_points', 'NUM_POINTS': {'train': 16384, 'test': 16384}}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}]
2023-06-22 10:24:04,828   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
2023-06-22 10:24:04,828   INFO  ----------- MODEL -----------
2023-06-22 10:24:04,828   INFO  cfg.MODEL.NAME: PointRCNN
2023-06-22 10:24:04,829   INFO  ----------- BACKBONE_3D -----------
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.NAME: PointNet2MSG
2023-06-22 10:24:04,829   INFO  ----------- SA_CONFIG -----------
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.NPOINTS: [4096, 1024, 256, 64]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.RADIUS: [[0.1, 0.5], [0.5, 1.0], [1.0, 2.0], [2.0, 4.0]]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.NSAMPLE: [[16, 32], [16, 32], [16, 32], [16, 32]]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.SA_CONFIG.MLPS: [[[16, 16, 32], [32, 32, 64]], [[64, 64, 128], [64, 96, 128]], [[128, 196, 256], [128, 196, 256]], [[256, 256, 512], [256, 384, 512]]]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.BACKBONE_3D.FP_MLPS: [[128, 128], [256, 256], [512, 512], [512, 512]]
2023-06-22 10:24:04,829   INFO  ----------- POINT_HEAD -----------
2023-06-22 10:24:04,829   INFO  cfg.MODEL.POINT_HEAD.NAME: PointHeadBox
2023-06-22 10:24:04,829   INFO  cfg.MODEL.POINT_HEAD.CLS_FC: [256, 256]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.POINT_HEAD.REG_FC: [256, 256]
2023-06-22 10:24:04,829   INFO  cfg.MODEL.POINT_HEAD.CLASS_AGNOSTIC: False
2023-06-22 10:24:04,829   INFO  cfg.MODEL.POINT_HEAD.USE_POINT_FEATURES_BEFORE_FUSION: False
[... 389 lines cut ...]
    return self._process_data(data)
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/dataloader.py", line 1371, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.10/dist-packages/torch/_utils.py", line 644, in reraise
    raise exception
ValueError: Caught ValueError in DataLoader worker process 0.
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/worker.py", line 308, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/fetch.py", line 51, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/fetch.py", line 51, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/content/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 425, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 185, in prepare_data
    data_dict = self.data_augmentor.forward(
  File "/content/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 302, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/augmentor/database_sampler.py", line 468, in __call__
    sampled_boxes = np.stack([x['box3d_lidar'] for x in sampled_dict], axis=0).astype(np.float32)
  File "<__array_function__ internals>", line 180, in stack
  File "/usr/local/lib/python3.10/dist-packages/numpy/core/shape_base.py", line 422, in stack
    raise ValueError('need at least one array to stack')
ValueError: need at least one array to stack

regenthor, 2023-07-18:
did you find any solution? @Choubey176

closed 2023-09-01 by github-actions[bot]

Petros626, 2024-10-18:
@Kazawaryu when using KITTI for training the filter rules should be fine? Except of course, that the rules don't fit custom data.