open-mmlab/OpenPCDet #1396: how to train not to load the ckpt before？

markwave, 2023-06-29:
When using openpcdet to train a model, I found that it automatically loads checkpoints from previous training sessions when I want to train the same model multiple times. Apart from manually moving the results of the previous training, is there a way to train the same model multiple times without loading the previous checkpoints and instead start training from scratch, and then save the results of each retraining session to a separate folder?

jihanyang, 2023-06-29:
You can: 1. use --extra_tag to distinguish different experiments. 2. delete all preivous ckpts.

closed 2023-06-29 by jihanyang

markwave, 2023-06-29:
> You can: 1. use --extra_tag to distinguish different experiments. or 2. delete all preivous ckpts.

thanks for your help，the first advice，Do I just add different extra_tag "--extra_tag" in the train command that runs different times on same model  can solve this question , And do not need to other changes in other files?

jihanyang, 2023-06-30:
Yes

markwave, 2023-06-30:
> Yes

thanks for your help, it helps me directly.