open-mmlab/OpenPCDet #1397: Processing the waymo dataset

ShengStar, 2023-07-01:
When I processed the waymo dataset through “python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos
    --cfg_file tools/cfgs/dataset_configs/waymo_dataset.yaml”, I found that the speed was particularly slow. How long does this process usually take for everyone.

chenshi3, 2023-07-01:
It depends on the number of cpu cores. Maybe you need 12 hours or more.

Haiyang-W, 2023-07-01:
It varies greatly in different machines, for example, SSD hard disk may only take a few hours.

ShengStar, 2023-07-01:
> It depends on the number of cpu cores. Maybe you need 12 hours or more.
Thank you very much for your help. Based on your suggestion, I have allocated 20 cores and 30GB of memory for this task. Do you think this is sufficient? In addition, during the processing, some warnings appeared, and I would like to ask if these affect the normal operation of the program.
“
2023-07-01 08:06:10.897263: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.898238: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.900426: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.900462: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.901601: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.905629: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.905667: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.906194: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.906232: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.906787: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.907333: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations:  AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
”
“
2023-07-01 08:06:15.174167: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.174527: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.231305: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.231665: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.307324: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.307678: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.318262: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.318791: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.348125: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.348762: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.374796: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.375157: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.408873: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.409219: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.427122: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.427502: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
”

ShengStar, 2023-07-01:
> It varies greatly in different machines, for example, SSD hard disk may only take a few hours.

Thank you very much for your help.  During the processing, some warnings appeared, and I would like to ask if these affect the normal operation of the program.
“
2023-07-01 08:06:10.897263: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations: AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.898238: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations: AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.900426: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.900462: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.901601: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations: AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.905629: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.905667: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.906194: E tensorflow/stream_executor/cuda/cuda_driver.cc:328] failed call to cuInit: CUDA_ERROR_NO_DEVICE: no CUDA-capable device is detected
2023-07-01 08:06:10.906232: I tensorflow/stream_executor/cuda/cuda_diagnostics.cc:163] no NVIDIA GPU device is present: /dev/nvidia0 does not exist
2023-07-01 08:06:10.906787: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations: AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
2023-07-01 08:06:10.907333: I tensorflow/core/platform/cpu_feature_guard.cc:142] This TensorFlow binary is optimized with oneAPI Deep Neural Network Library (oneDNN) to use the following CPU instructions in performance-critical operations: AVX2 AVX512F FMA
To enable them in other operations, rebuild TensorFlow with the appropriate compiler flags.
”
“
2023-07-01 08:06:15.174167: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.174527: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.231305: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.231665: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.307324: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.307678: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.318262: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.318791: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.348125: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.348762: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.374796: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.375157: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.408873: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.409219: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
2023-07-01 08:06:15.427122: I tensorflow/compiler/mlir/mlir_graph_optimization_pass.cc:176] None of the MLIR Optimization Passes are enabled (registered 2)
2023-07-01 08:06:15.427502: I tensorflow/core/platform/profile_utils/cpu_utils.cc:114] CPU Frequency: 2600000000 Hz
”

closed 2023-08-16 by github-actions[bot]

YuRui-Learning, 2024-01-11:
ello, when i run test or train function , i will also meet this question in eval. It is some issue aboud tensorflow ?
Traceback (most recent call last): File "eval.py", line 207, in <module> main() File "eval.py", line 203, in main eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test) File "eval.py", line 65, in eval_single_ckpt eval_result.eval_one_epoch( File "/root/MSF/tools/eval_utils/eval_result.py", line 62, in eval_one_epoch result_str, result_dict = dataset.evaluation( File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 476, in evaluation ap_result_str, ap_dict = waymo_eval(eval_det_annos, eval_gt_annos) File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_dataset.py", line 459, in waymo_eval ap_dict = eval.waymo_evaluation( File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 211, in waymo_evaluation metrics = self.build_graph(graph) File "/root/MSF/tools/../pcdet/datasets/waymo/waymo_eval.py", line 127, in build_graph metrics = detection_metrics.get_detection_metric_ops( File "/root/.local/lib/python3.8/site-packages/waymo_open_dataset/metrics/python/detection_metrics.py", line 134, in get_detection_metric_ops ground_truth_speed = tf.zeros((num_gt_boxes, 2), tf.float32) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/util/dispatch.py", line 206, in wrapper return target(*args, **kwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2912, in wrapped tensor = fun(*args, **kwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2962, in zeros output = _constant_if_small(zero, shape, dtype, name) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/ops/array_ops.py", line 2897, in _constant_if_small if np.prod(shape) < 1000: File "<__array_function__ internals>", line 200, in prod File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 3076, in prod return _wrapreduction(a, np.multiply, 'prod', axis, dtype, out, File "/usr/local/lib/python3.8/dist-packages/numpy/core/fromnumeric.py", line 86, in _wrapreduction return ufunc.reduce(obj, axis, dtype, out, **passkwargs) File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/ops.py", line 867, in __array__ raise NotImplementedError( NotImplementedError: Cannot convert a symbolic Tensor (strided_slice:0) to a numpy array. This error may indicate that you're trying to pass a Tensor to a NumPy call, which is not supported eval:   0%|          | 0/4931 [02:31<?, ?it/s]

RhythmOfTheRain-Byte, 2024-03-03:
Excuse me. How to solve the problem as blow.
When i run python -m pcdet.datasets.waymo.waymo_dataset --func create_waymo_infos  --cfg_file tools/cfgs/dataset_configs/waymo_dataset.yaml

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/process.py", line 239, in _process_worker
    r = call_item.fn(*call_item.args, **call_item.kwargs)
  File "/usr/lib/python3.8/concurrent/futures/process.py", line 198, in _process_chunk
    return [fn(*args) for args in chunk]
  File "/usr/lib/python3.8/concurrent/futures/process.py", line 198, in <listcomp>
    return [fn(*args) for args in chunk]
  File "/home/mlcnet/pcdet/datasets/waymo/waymo_utils.py", line 196, in process_single_sequence
    dataset = tf.data.TFRecordDataset(str(sequence_file), compression_type='')
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/data/ops/readers.py", line 336, in __init__
    filenames = _create_or_validate_filenames_dataset(filenames)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/data/ops/readers.py", line 56, in _create_or_validate_filenames_dataset
    filenames = ops.convert_to_tensor(filenames, dtype_hint=dtypes.string)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/ops.py", line 1328, in convert_to_tensor
    ret = conversion_func(
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/constant_op.py", line 321, in _constant_tensor_conversion_function
    return constant(v, dtype=dtype, name=name)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/constant_op.py", line 261, in constant
    return _constant_impl(value, dtype, shape, name, verify_shape=False,
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/constant_op.py", line 270, in _constant_impl
    t = convert_to_eager_tensor(value, ctx, dtype)
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/framework/constant_op.py", line 95, in convert_to_eager_tensor
    ctx.ensure_initialized()
  File "/usr/local/lib/python3.8/dist-packages/tensorflow/python/eager/context.py", line 515, in ensure_initialized
    context_handle = pywrap_tfe.TFE_NewContext(opts)
tensorflow.python.framework.errors_impl.InternalError: CUDA runtime implicit initialization on GPU:0 failed. Status: out of memory
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/mlcnet/pcdet/datasets/waymo/waymo_dataset.py", line 416, in <module>
    create_waymo_infos(
  File "/home/mlcnet/pcdet/datasets/waymo/waymo_dataset.py", line 375, in create_waymo_infos
    waymo_infos_train = dataset.get_infos(
  File "/home/mlcnet/pcdet/datasets/waymo/waymo_dataset.py", line 113, in get_infos
    sequence_infos = list(tqdm(executor.map(process_single_sequence, sample_sequence_file_list),
  File "/usr/local/lib/python3.8/dist-packages/tqdm/std.py", line 1129, in __iter__
    for obj in iterable:
  File "/usr/lib/python3.8/concurrent/futures/process.py", line 484, in _chain_from_iterable_of_lists
    for element in iterable:
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 611, in result_iterator
    yield fs.pop().result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 439, in result
    return self.__get_result()
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 388, in __get_result
    raise self._exception
tensorflow.python.framework.errors_impl.InternalError: CUDA runtime implicit initialization on GPU:0 failed. Status: out of memory