open-mmlab/OpenPCDet #1399: MPPNet with custom dataset

pateljayam, 2023-07-03:
Hello Authors

Thank you for the contribution to the scientific community. 

I am playing with OpenPCDet to run with custom dataset (a sequence of point clouds and robot pose) and I want to try training of MPPNet with it.
I do not have range image in my custom dataset, and was wondering if I can alter the model to run without the "elongation" feature of the lidar points. 
Would it still work? 
Do I have to consider any other change?

Please let me know at your convenience. 

Regards
Jayam

Tottowich, 2023-07-16:
@pateljayam Hi! Have you had any success/ made any progress in your exploration? I'm also looking to try some multi-frame models and like you, I'm also looking into using it on a custom dataset.

pateljayam, 2023-07-24:
Hello @Tottowich 
I have been slowly digging into the methodology and made some progress. 
I removed the elongation feature and only use ['x', 'y', 'z', 'intensity', 'timestamp']
For CenterPoint you will not have to make any changes in the network. 
But, for MPPNet you might have to make changes. #1147, this issue was very helpful. 
Happy training. :)
All the best.

closed 2023-08-24 by pateljayam