open-mmlab/OpenPCDet #1400: ImportError

learnuser1, 2023-07-04:
when i run the commend of python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos \
    --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml --version v1.0-trainval.
ImportError: cannot import name 'roiaware_pool3d_cuda' from 'pcdet.ops.roiaware_pool3d' (/home/data/OpenPCDet/pcdet/ops/roiaware_pool3d/__init__.py), and i can't find the function's position of roiaware_pool3d_cuda,
how to deal with this problem

BBBeatrice, 2023-07-14:
Did you solve it? I encounter the same problem.

learnuser1, 2023-07-19:
> 

yes, i solved it.just run the commend ,the system can generate the  roiaware_pool3d
python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos \
    --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml \
    --version v1.0-trainval

closed 2023-09-02 by github-actions[bot]