open-mmlab/OpenPCDet #1403: How to calculate model's FLOPs?

M0NSTERCOCO, 2023-07-05:
Every paper which uses OpenPCDet Framework calculates FLOPs or FPS, but there doesn't have any tools in this framework to compute this direactly. I have used torchsummary, torchstat and many other tools, they didn't work and had error below.
`Traceback (most recent call last):
  File "test.py", line 229, in <module>
    main()
  File "test.py", line 209, in main
    summary(model, (3,8,4))
  File "/opt/conda/lib/python3.7/site-packages/torchsummary/torchsummary.py", line 72, in summary
    model(*x)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "../pcdet/models/detectors/IASSD.py", line 13, in forward
    batch_dict = cur_module(batch_dict)
  File "/opt/conda/lib/python3.7/site-packages/torch/nn/modules/module.py", line 1120, in _call_impl
    result = forward_call(*input, **kwargs)
  File "../pcdet/models/backbones_3d/IASSD_backbone.py", line 143, in forward
    batch_size = batch_dict['batch_size']
IndexError: too many indices for tensor of dimension 4`

Sunmmming, 2023-07-24:
I also encountered the same problem. Have you solved this problem yet?

jihanyang, 2023-07-24:
You can refer to our other project https://github.com/CVMI-Lab/SparseKD/blob/master/docs/GETTING_STARTED.md to check out how to calculate model metrics in openpcdet.

Sunmmming, 2023-07-25:
Thank you for your help, but there are still some errors . Refer to  https://github.com/CVMI-Lab/SparseKD/blob/master/docs/GETTING_STARTED.md, I installed Thop and spconv==2.1.20, and change the conv
`cp extra_files/conv.py ${CONDA_PATH}/envs/${ENV_NAME}/lib/${PYTHON_VERSION}/site-packages/spconv/pytorch/`. 
 Also change some codes.  
cuda 10.2 torch 1.7.0
1. Comment out m.total_acts +=y.features.shape[0]*y.features.shape[1] in extra_files/conv.py
    Because "torch.nn.modules.module.ModuleAttributeError: 'SubMConv3d'" object has not attribute 'total_acts'"
```
@staticmethod
def count_your_model(m,x,y):
    m.total_ops +=torch.DoubleTensor([int(y.flops)])
    #m.total_acts +=y.features.shape[0]*y.features.shape[1] 
```
3. change cal_fops function in pcdet/utils/common_utils/cal_flops(model, batch_dict)
ori:
```
def cal_flops(model,batch_dict):
    mac, params,acts = profile_acts(model, inputs=(batch_dict,),
                                    custon_ops ={
                                            spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
                                            spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
                                            spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
                                            spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
}
```
cahnged:
```
def cal_flops(model,batch_dict):
    mac, params= profile(model, inputs=(batch_dict,),
                                    custon_ops ={
                                            spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
                                            spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
                                            spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
                                            spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
}
)
```
And then use the model ,https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml and follow code to get the model's Flops 
```
    mac, params= profile(model, inputs=(batch_dict,),
                                    custon_ops ={
                                            spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
                                            spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
                                            spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
                                            spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
   macs2,params2 = clever_format([macs1,parasm1],"%0.3f")
   print('macs1=',macs1)
   print('params1=',params1)
   print('macs2=',macs2)
   print('params2=',params2)
```
And, the reuslts if follow:
```
macs1 = 68317175646786.0
params1 = 7981398.0
macs2 = 68.317T
params2 = 7.981M
```
The macs is so huge , is there anything wrong？Wish for you help.

Sunmmming, 2023-07-25:
I Fould in the extra_files/conv.py , The method of calculating floats is incorrect！
ori:  [extra_files/conv.py](https://github.com/CVMI-Lab/SparseKD/blob/master/tools/extra_files/conv.py) #233
```
 if self.conv1x1:
      if FILTER_HWIO:
           features = torch.mm(
                  input.features,
                    self.weight.view(self.out_channels, self.in_channels).T)
       else:
           features = torch.mm(
                    input.features,
                    self.weight.view(self.in_channels, self.out_channels))

     flops = input.features.shape[0] * input.features.shape[1] * features.shape[1] * features.shape[0]
```
FLOPS for torch.mm  should: input.features.shape[0] * input.features.shape[1] * features.shape[1]

jihanyang, 2023-07-26:
@Sunmmming  Thank you for pointing this out, I will correct it soon!

closed 2023-09-09 by github-actions[bot]

mystorm16, 2024-01-26:
> Thank you for your help, but there are still some errors . Refer to https://github.com/CVMI-Lab/SparseKD/blob/master/docs/GETTING_STARTED.md, I installed Thop and spconv==2.1.20, and change the conv `cp extra_files/conv.py ${CONDA_PATH}/envs/${ENV_NAME}/lib/${PYTHON_VERSION}/site-packages/spconv/pytorch/`. Also change some codes. cuda 10.2 torch 1.7.0
> 
> 1. Comment out m.total_acts +=y.features.shape[0]*y.features.shape[1] in extra_files/conv.py
>    Because "torch.nn.modules.module.ModuleAttributeError: 'SubMConv3d'" object has not attribute 'total_acts'"
> 
> ```
> @staticmethod
> def count_your_model(m,x,y):
>     m.total_ops +=torch.DoubleTensor([int(y.flops)])
>     #m.total_acts +=y.features.shape[0]*y.features.shape[1] 
> ```
> 
> 3. change cal_fops function in pcdet/utils/common_utils/cal_flops(model, batch_dict)
>    ori:
> 
> ```
> def cal_flops(model,batch_dict):
>     mac, params,acts = profile_acts(model, inputs=(batch_dict,),
>                                     custon_ops ={
>                                             spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
>                                             spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
>                                             spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
>                                             spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
> }
> ```
> 
> cahnged:
> 
> ```
> def cal_flops(model,batch_dict):
>     mac, params= profile(model, inputs=(batch_dict,),
>                                     custon_ops ={
>                                             spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
>                                             spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
>                                             spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
>                                             spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
> }
> )
> ```
> 
> And then use the model ,https://github.com/open-mmlab/OpenPCDet/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml and follow code to get the model's Flops
> 
> ```
>     mac, params= profile(model, inputs=(batch_dict,),
>                                     custon_ops ={
>                                             spconv.SubMConv3D: spconv.SubMConv3D.count_your_model,
>                                             spconv.SparseConv3D: spconv.SparseConv3D.count_your_model,
>                                             spconv.SubMConv2D: spconv.SubMConv2D.count_your_model,
>                                             spconv.SparseConv2D: spconv.SparseConv2D.count_your_model,
>    macs2,params2 = clever_format([macs1,parasm1],"%0.3f")
>    print('macs1=',macs1)
>    print('params1=',params1)
>    print('macs2=',macs2)
>    print('params2=',params2)
> ```
> 
> And, the reuslts if follow:
> 
> ```
> macs1 = 68317175646786.0
> params1 = 7981398.0
> macs2 = 68.317T
> params2 = 7.981M
> 
> SubMConv2d()
> ops: 1868098502656.0 params:16512.0
> ```
> 
> The macs is so huge , is there anything wrong？Wish for you help.

THX for this method~   a small mistake: custon_ops->custom_ops

SparisWeir, 2025-08-26:
@jihanyang Hello, I also want to calculate flops for a model based on openPCDet, I downloaded that project by following the link you gave, but I don't know how to calculate my own model with your code。Wish for your help