open-mmlab/OpenPCDet #1407: RuntimeError: Address already in use

1800710138, 2023-07-14:
I want to train multiple GPUs and execute the following code, but an error occurred:
python -m torch.distributed.launch --nproc_per_node=2 train.py --launcher pytorch --cfg_file cfgs/kitti_models/pv_rcnn.yaml --tcp_port 35563
The port number is functioning properly：
[image]
But he reported the error

jihanyang, 2023-07-15:
just re-run it again, it will find an unused port automatically.

1800710138, 2023-07-15:
I rerun the code, but it still doesn't work
[image]

1800710138, 2023-07-15:
> just re-run it, it will find an unused port automatically.

I rerun the code, but it still doesn't work
[image]

jihanyang, 2023-07-26:
Would you like to use our provided script to run the code? Since we provide the scripts to auto detect the free port

closed 2023-09-09 by github-actions[bot]