open-mmlab/OpenPCDet #1409: NCCL stuck at collective operations timeout

OuYaozhong, 2023-07-18:
Hi, @sshaoshuai @chenshi3 @jihanyang @yukang2017 @djiajunustc @Gus-Guo @Cedarch @acivgin1 

I have stuck at the NCCL problem.
If i use multiple GPUs at single machine, it comes at a NCCL communication error: `[Rank 1] Watchdog caught collective operation timeout:`

**Location:**
The location that stuck and raise timeout error, is random.
But, most of time is located at the `dist.all_reduce()` or some `xx.alltogether`.
It seems as the NCCL communication problem.

**Special Features:**
If the problem run normally, the utilization of GPU will change with time.
But if the program stuck at this kind of problem, the utilization of all used GPUs are 100%, till the timeout error raised.

**Some Trial:**
I have do some trial to find what happen and help your developer to save time to locate the problem.
The most meaningful finding is telling below:

At [pcdet/utils/common_utils.py](https://github.com/open-mmlab/OpenPCDet/blob/557d463793f6ff63f0301e851a047d29924fb1fe/pcdet/utils/common_utils.py#L190-L191), we have the code:
```
if mp.get_start_method(allow_none=True) is None:
        mp.set_start_method('spawn')
```

If delete these two line code, it can run normally without any NCCL timeout error for several hours.
And if use `torch.multiprocessing`, the NCCL timeout error will be met within 1 hours in general.

And,
If ***DISABLE*** the `torch.multiprocessing`, **the ETA time of program reach 2x higher** (may be relative to the number of GPU as i am using 2). And only 43G memory will be used instead of about 133G will used if ***ENABLE*** `torch.multiprocessing`.

_____________________________
**In conculsion, the problem seems relative to the mixed use of both `torchrun` and `torch.multiprocessing`. 
The mixed use will cause higher memory usage, and reach higher speed, but will face NCCL error `collective operations timeout` at random within short time.
Only use `torchrun` seems get rid of the NCCL timeout error, but run slower, and use much less memory.**
_____________________________

Below are some detailed log about this issue.
**Environment:**
```
Tue Jul 18 20:06:39 2023       
+---------------------------------------------------------------------------------------+
| NVIDIA-SMI 535.54.03              Driver Version: 535.54.03    CUDA Version: 12.2     |
|-----------------------------------------+----------------------+----------------------+
| GPU  Name                 Persistence-M | Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp   Perf          Pwr:Usage/Cap |         Memory-Usage | GPU-Util  Compute M. |
|                                         |                      |               MIG M. |
|=========================================+======================+======================|
|   0  NVIDIA GeForce RTX 3090        Off | 00000000:67:00.0 Off |                  N/A |
| 62%   54C    P2             142W / 350W |  10891MiB / 24576MiB |    100%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
|   1  NVIDIA GeForce RTX 3090        Off | 00000000:68:00.0 Off |                  N/A |
| 81%   59C    P2             156W / 350W |  10816MiB / 24576MiB |    100%      Default |
|                                         |                      |                  N/A |
+-----------------------------------------+----------------------+----------------------+
                                                                                         
+---------------------------------------------------------------------------------------+
| Processes:                                                                            |
|  GPU   GI   CI        PID   Type   Process name                            GPU Memory |
|        ID   ID                                                             Usage      |
|=======================================================================================|
|    0   N/A  N/A      1644      G   /usr/lib/xorg/Xorg                            4MiB |
|    0   N/A  N/A   1890568      C   ........./.conda/envs/pcdet/bin/python    10874MiB |
|    1   N/A  N/A      1644      G   /usr/lib/xorg/Xorg                           14MiB |
|    1   N/A  N/A   1890569      C   ........./.conda/envs/pcdet/bin/python    10788MiB |
+---------------------------------------------------------------------------------------+
```

```
$ conda list | grep torch
[... 710 lines cut ...]
    run(args)
  File "/home/me/.conda/envs/pcdet/lib/python3.11/site-packages/torch/distributed/run.py", line 785, in run
    elastic_launch(
  File "/home/me/.conda/envs/pcdet/lib/python3.11/site-packages/torch/distributed/launcher/api.py", line 134, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/me/.conda/envs/pcdet/lib/python3.11/site-packages/torch/distributed/launcher/api.py", line 250, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError:
========================================================
train.py FAILED
--------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
--------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-07-18_20:23:52
  host      : AI-3090
  rank      : 1 (local_rank: 1)
  exitcode  : -6 (pid: 1890569)
  error_file: <N/A>
  traceback : Signal 6 (SIGABRT) received by PID 1890569
========================================================
```

chenshi3, 2023-07-19:
Have you ever monitored your machine's memory usage? This could potentially be caused by a memory failure.

OuYaozhong, 2023-07-19:
> Have you ever monitored your machine's memory usage? This could potentially be caused by a memory failure.

Hi, @chenshi3 

It seems that not the out of memory problem. The total size of memory of my machine is up to 256G, and though have not monitor it, the maximal usage of memory when it ran with `torch.multiprocessing` is around 133G, which is only around half of the total.

And the GPU memory is about 24G, and occupies about 10G per one.

OuYaozhong, 2023-07-19:
@chenshi3 And i have a problem that, is it necessary that use `torch.multiprocessing` to start multiprocessing ?

I see that in many DDP best practice, the mp.set_start_method() is an option, and not all of them are used.

And in my case, set the mp.set_start_method('spawn') indeed make the program run 2x faster.

Is there any reason ?

closed 2023-09-02 by github-actions[bot]

PoPdark, 2025-07-24:
@OuYaozhong Hi, have you solved this problem, i have the same trouble as you.