open-mmlab/OpenPCDet #1413: Training and testing TransFusion-L with openpcdet version 0.2.0 (ST3D)

Galaxy-ZRX, 2023-07-24:
Hi, I am currently using the pcdet 2.0 which is from the ST3D repo, may I ask if it's possible to add transfusion-L to it?

I also noticed thatit has been supported by the latest pcdet, and I am not sure if I can simply create a new environment with the latest pcdet (without the need of re-preparing the datasets) so that I can use transfusion-L there?

Many thanks

chenshi3, 2023-07-25:
Yep. You just need to copy and paste the related code of transfusion-L.

Galaxy-ZRX, 2023-07-25:
Thank you! I will try soon.

chenshi3, 2023-07-25:
> May I confirm that I need to change these codes as you committed? [4dc1849](https://github.com/open-mmlab/OpenPCDet/commit/4dc18496f09154eb4d4babd9f2dd90f35bffc6bf)

Certainly, to begin, you can test with the provided checkpoints.

Galaxy-ZRX, 2023-07-26:
Hi @chenshi3 , I have firstly tried to reproduce the results of TransFusion-L on current pcdet version (0.6.0), however my results are quite different from the reported ones:

2023-07-26 10:58:18,835   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.20, 0.18, 0.21, 0.43, 0.22 | 67.56, 78.67, 82.59, 85.03 | mean AP: 0.7846314779117628
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.40, 0.24, 0.26, 0.45, 0.26 | 25.21, 44.90, 53.47, 57.96 | mean AP: 0.45383654553284947
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.75, 0.45, 1.16, 0.14, 0.34 | 0.49, 7.56, 16.16, 27.64 | mean AP: 0.12961399733232265
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.43, 0.23, 0.24, 1.28, 0.28 | 28.09, 54.43, 69.15, 72.58 | mean AP: 0.5606204998406492
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.63, 0.26, 1.06, 0.46, 0.23 | 4.13, 20.90, 35.82, 45.82 | mean AP: 0.26669573720829554
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.35, 0.29, 0.12, nan, nan | 33.25, 52.40, 60.15, 62.77 | mean AP: 0.5214156825794469
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.23, 0.26, 0.55, 1.22, 0.21 | 33.32, 40.71, 42.72, 46.18 | mean AP: 0.40736125116755956
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.19, 0.27, 0.69, 0.43, 0.01 | 25.16, 26.90, 27.53, 28.58 | mean AP: 0.2704486481612039
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.32, 0.29, 1.37, 0.83, 0.23 | 53.19, 61.95, 64.88, 67.60 | mean AP: 0.6190817176746111
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.18, 0.37, nan, nan, nan | 54.19, 55.85, 58.78, 63.03 | mean AP: 0.5796267687608527
--------------average performance-------------
trans_err:	 0.3669
scale_err:	 0.2831
orient_err:	 0.6275
vel_err:	 0.6534
attr_err:	 0.2234
mAP:	 0.4593
NDS:	 0.5142

mAP: 0.4593
mATE: 0.3669
mASE: 0.2831
mAOE: 0.6275
mAVE: 0.6534
mAAE: 0.2234
NDS: 0.5142

I train the model with batch size = 4 and lr = 5e-5 for 20 epochs, following the settings from original transfusion (they use batch size = 2*8 and lr = 1e-4).

Could you please check if such settings are suitable?

chenshi3, 2023-07-26:
You can begin by testing with the provided checkpoints to validate the correctness of code and dataset. The results are obtained using a batch size of 4*8 and a learning rate of 1e-3. For your setting, I recommend adjusting the lr to 1e-4 and decreasing the number of epochs in [DisableAugmentationHook](https://github.com/open-mmlab/OpenPCDet/blob/557d463793f6ff63f0301e851a047d29924fb1fe/tools/cfgs/nuscenes_models/transfusion_lidar.yaml#L171).

Galaxy-ZRX, 2023-07-29:
Hi @chenshi3 , I modified the lr to 2.5e-4 with batch size = 8, and also reduce the disableAugmentationHook epoch to 3. 

The performance of epoch 20 is improved a bit (from 45.9 to 52.6),  but still far from the reported ones. I have tested the provided checkpoint which achieves 64.55 AP, indicating that the code and dataset are without problems. Do you have any suggestions on this? Thank you!

Galaxy-ZRX, 2023-07-29:
I also disabled the balanced_resampling in nuscenes_dataset.yaml to save time, but I am not sure if such change will cause the heavy performance drop.

chenshi3, 2023-07-29:
> sabled the balanced_resampling in nuscenes_dataset.yaml to save time, but I am not sure if such change will cause the heavy performance d

Firstly,  the balanced_resampling in nuscenes_dataset.yaml significantly impacts performance to performance. Secondly, I recommend reducing the disableAugmentationHook epoch to 1 or 2 for better results..

Galaxy-ZRX, 2023-08-01:
Hi @chenshi3 , thank you for your help and I have implemented transfusion on pcdet2.0. However, I only want to train and test with car class instead of the 10 classes in nuScenes. Do you have any suggestions on modifying the config file based on this? For example, should I reduce the number of proposals from 200 to 20?

closed 2023-09-16 by github-actions[bot]