open-mmlab/OpenPCDet #1414: Expected a value of type 'float' for argument 'p' but instead found type 'int'.

REFRAING, 2023-07-26:
when I do 
python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos \
>     --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml \
>     --version v1.0-trainval \
>     --with_cam

Traceback (most recent call last):
  File "/root/miniconda3/lib/python3.8/runpy.py", line 185, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/root/miniconda3/lib/python3.8/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/root/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script
    fn = torch._C._jit_script_compile(
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_recursive.py", line 838, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script
    fn = torch._C._jit_script_compile(
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_recursive.py", line 838, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script
    fn = torch._C._jit_script_compile(
RuntimeError: 

normalize(Tensor input, float p=2., int dim=1, float eps=9.9999999999999998e-13, Tensor? out=None) -> (Tensor):
Expected a value of type 'float' for argument 'p' but instead found type 'int'.
:
  File "/root/miniconda3/lib/python3.8/site-packages/kornia/geometry/conversions.py", line 346
            "Input must be a tensor of shape (*, 4). Got {}".format(
                quaternion.shape))
    return F.normalize(quaternion, p=2, dim=-1, eps=eps)
           ~~~~~~~~~~~ <--- HERE
'normalize_quaternion' is being compiled since it was called from 'quaternion_to_rotation_matrix'
  File "/root/miniconda3/lib/python3.8/site-packages/kornia/geometry/conversions.py", line 380
                quaternion.shape))
    # normalize the input quaternion
    quaternion_norm: torch.Tensor = normalize_quaternion(quaternion)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE

    # unpack the normalized quaternion components
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )

neverstoplearn, 2023-08-08:
> when I do python -m pcdet.datasets.nuscenes.nuscenes_dataset --func create_nuscenes_infos \
> 
> > ```
> > --cfg_file tools/cfgs/dataset_configs/nuscenes_dataset.yaml \
> > --version v1.0-trainval \
> > --with_cam
> > ```
> 
> Traceback (most recent call last): File "/root/miniconda3/lib/python3.8/runpy.py", line 185, in _run_module_as_main mod_name, mod_spec, code = _get_module_details(mod_name, _Error) File "/root/miniconda3/lib/python3.8/runpy.py", line 111, in _get_module_details **import**(pkg_name) File "/root/OpenPCDet/pcdet/datasets/**init**.py", line 15, in from .argo2.argo2_dataset import Argo2Dataset File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in def quat_to_mat(quat_wxyz: Tensor) -> Tensor: File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script fn = torch._C._jit_script_compile( File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_recursive.py", line 838, in try_compile_fn return torch.jit.script(fn, _rcb=rcb) File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script fn = torch._C._jit_script_compile( File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_recursive.py", line 838, in try_compile_fn return torch.jit.script(fn, _rcb=rcb) File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1310, in script fn = torch._C._jit_script_compile( RuntimeError:
> 
> normalize(Tensor input, float p=2., int dim=1, float eps=9.9999999999999998e-13, Tensor? out=None) -> (Tensor): Expected a value of type 'float' for argument 'p' but instead found type 'int'. : File "/root/miniconda3/lib/python3.8/site-packages/kornia/geometry/conversions.py", line 346 "Input must be a tensor of shape (*, 4). Got {}".format( quaternion.shape)) return F.normalize(quaternion, p=2, dim=-1, eps=eps) ~~~~~~~~~~~ <--- HERE 'normalize_quaternion' is being compiled since it was called from 'quaternion_to_rotation_matrix' File "/root/miniconda3/lib/python3.8/site-packages/kornia/geometry/conversions.py", line 380 quaternion.shape)) # normalize the input quaternion quaternion_norm: torch.Tensor = normalize_quaternion(quaternion) ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
> 
> ```
> # unpack the normalized quaternion components
> ```
> 
> 'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat' File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19 (...,3,3) 3D rotation matrices. """ return C.quaternion_to_rotation_matrix( ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE )

have you fix the error,I meet the same error?

OuYaozhong, 2023-08-29:
The same problem

GongDianCheng, 2023-09-01:
> 
Thank you！
@shijianjian

yang-jiaye, 2023-09-21:
if you are just using nuscenes dataset, you can modify /pcdet/dataset/init,py, commenting the following line