open-mmlab/OpenPCDet #1416: SecondNet Custom Data Training Maximum Recursion Depth

regenthor, 2023-07-27:
I want to train a new model on second net with my own dataset.

I prepared the data and created my dataset with this command.
"python -m pcdet.datasets.custom.custom_dataset create_custom_infos tools/cfgs/dataset_configs/custom_dataset.yaml"

After the dataset was created, I ran the train command.
"python train.py --cfg_file cfgs/custom_models/second.yaml  --pretrained_model second_7862.pth --epochs 2   --use_amp"

### **I get this error
RecursionError: maximum recursion depth exceeded while calling a Python object**

**Point cloud range, z min = -14, z max = 0, voxel_size = 0.15
(z_max - z_min) / voxel_size = I know it should not exceed 40, what can I do about this? This is how the Z values come from the lidar. Any chance of changing the voxel_size?
POINT_CLOUD_RANGE: [-3.8, -3.0, -14, 2.6, 1.8, 0]
VOXEL_SIZE: [0.1, 0.1, 0.15]**

### **ERROR LOGS**

2023-07-26 13:08:32,875   INFO  **********************Start training custom_models/second(default)**********************
epochs:   0% 0/2 [00:00<?, ?it/s]
epochs:   0% 0/2 [00:18<?, ?it/s]
Traceback (most recent call last):
  File "/content/OpenPCDet/tools/train.py", line 230, in <module>
    main()
  File "/content/OpenPCDet/tools/train.py", line 175, in main
    train_model(
  File "/content/OpenPCDet/tools/train_utils/train_utils.py", line 180, in train_model
    accumulated_iter = train_one_epoch(
  File "/content/OpenPCDet/tools/train_utils/train_utils.py", line 33, in train_one_epoch
    batch = next(dataloader_iter)
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/dataloader.py", line 634, in __next__
    data = self._next_data()
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/dataloader.py", line 1346, in _next_data
    return self._process_data(data)
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/dataloader.py", line 1372, in _process_data
    data.reraise()
  File "/usr/local/lib/python3.10/dist-packages/torch/_utils.py", line 644, in reraise
    raise exception
RecursionError: Caught RecursionError in DataLoader worker process 0.
..
..
..
Original Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/worker.py", line 308, in _worker_loop
    data = fetcher.fetch(index)
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/fetch.py", line 51, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/usr/local/lib/python3.10/dist-packages/torch/utils/data/_utils/fetch.py", line 51, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)

File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 214, in prepare_data
    return self.__getitem__(new_index)
  File "/content/OpenPCDet/tools/../pcdet/datasets/custom/custom_dataset.py", line 109, in __getitem__
    data_dict = self.prepare_data(data_dict=input_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 185, in prepare_data
    data_dict = self.data_augmentor.forward(
  File "/content/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 302, in forward
    data_dict = cur_augmentor(data_dict=data_dict)
  File "/content/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 62, in random_world_flip
    gt_boxes, points, enable = getattr(augmentor_utils, 'random_flip_along_%s' % cur_axis)(
  File "/content/OpenPCDet/tools/../pcdet/datasets/augmentor/augmentor_utils.py", line 16, in random_flip_along_x
    enable = np.random.choice([False, True], replace=False, p=[0.5, 0.5])
  File "mtrand.pyx", line 981, in numpy.random.mtrand.RandomState.choice
  File "<__array_function__ internals>", line 180, in cumsum
  File "/usr/local/lib/python3.10/dist-packages/numpy/core/fromnumeric.py", line 2569, in cumsum
    return _wrapfunc(a, 'cumsum', axis=axis, dtype=dtype, out=out)
  File "/usr/local/lib/python3.10/dist-packages/numpy/core/fromnumeric.py", line 52, in _wrapfunc
    bound = getattr(obj, method, None)
RecursionError: maximum recursion depth exceeded while calling a Python object

regenthor, 2023-08-14:
any help?

lijin123red, 2023-09-25:
You can print your point cloud data and change your point cloud range.I change the range in z-axis then it solved.

closed 2023-11-10 by github-actions[bot]

llz666ha, 2024-10-27:
Is it solved, please