open-mmlab/OpenPCDet #1417: How to evaluate FPS for Voxel-RCNN ?

ammaryasirnaich, 2023-07-27:
Hi, I am trying to reproduce the FPS for the Voxel-RCNN but I don't see anything relevant metric in the test.py for that.

I have tried to write a script for that which gives me approx 23.5 fps with batch_size=1, which is below the official results(25 FPS). Would appreciate it if anyone can guide me!

```

def eval_one_epoch(cfg, args, model, dataloader, epoch_id, logger, dist_test=False, result_dir=None):
    result_dir.mkdir(parents=True, exist_ok=True)

    log_interval =50
    samples=2000

    
    dataset = dataloader.dataset
    class_names = dataset.class_names
    det_annos = []
 
    logger.info('*************** FPS %s EVALUATION *****************' % epoch_id)
    model.eval()

    if cfg.LOCAL_RANK == 0:
        progress_bar = tqdm.tqdm(total=len(dataloader), leave=True, desc='eval', dynamic_ncols=True)

     # the first several iterations may be very slow so skip them
    num_warmup = 5
    pure_inf_time = 0

    for i, batch_dict in enumerate(dataloader):
        load_data_to_gpu(batch_dict)

        torch.cuda.synchronize()
        start_time_fps = time.perf_counter()

        with torch.no_grad():
            pred_dicts, ret_dict = model(batch_dict)

        torch.cuda.synchronize()
        elapsed = time.perf_counter() - start_time_fps

        if i >= num_warmup:
            pure_inf_time += elapsed
            if (i + 1) % log_interval == 0:
                fps = (i + 1 - num_warmup) / pure_inf_time
                print(f'Done sample [{i + 1:<3}/ {samples}], '
                      f'fps: {fps:.1f} sample / s')

        if (i + 1) == samples:
            pure_inf_time += elapsed
            fps = (i + 1 - num_warmup) / pure_inf_time
            print(f'Overall fps: {fps:.1f} sample / s')
            break

        

if __name__ == '__main__':
    pass

```

yinheyanxian, 2023-08-29:
Hello, I also encountered this issue while evaluating Voxel R-CNN. Have you resolved it?

ammaryasirnaich, 2023-08-30:
> Hello, I also encountered this issue while evaluating Voxel R-CNN. Have you resolved it?

Actually, i used the shared code you calculate the FPS, although it doesn't match the claimed results.

yinheyanxian, 2023-08-31:
> 
I noticed that there is a Generate label finished (sec_per_example: 0.0168 second) in the automatic evaluation results after the model training is completed, which should be FPS

yinheyanxian, 2023-08-31:
> > Hello, I also encountered this issue while evaluating Voxel R-CNN. Have you resolved it?
> 
> Actually, i used the shared code you calculate the FPS, although it doesn't match the claimed results.

I noticed that there is a Generate label finished (sec_per_example: 0.0168 second) in the automatic evaluation results after the model training is completed, which should be FPS

closed 2023-10-16 by github-actions[bot]