open-mmlab/OpenPCDet #1419: The reproduction is inconsistent with Kitti's official website

JunjieWang0528, 2023-07-29:
When reproducing pv rcnn, I used the official configuration file and a GPU to obtain the same test set as the following results.
[image]
May I ask how to get another higher result on the leaderboard, as shown in the following figure.
[image]

tyutlk, 2023-07-31:
I think when we talking benchmark,we are talking about validation set,not test set.

JunjieWang0528, 2023-07-31:
> I think when we talking benchmark reproduced by openpcdet,we are talking about validation set,not test set.

Because I made improvements to PV-RCNN, there was an improvement in the validation set. The results obtained on the test set were better than the first screenshot, but not as good as the second screenshot. Therefore, I want to know where the problem may be.

JunjieWang0528, 2023-07-31:
> I think when we talking benchmark reproduced by openpcdet,we are talking about validation set,not test set.

Thank you for your reply. I know where the problem lies

closed 2023-07-31 by JunjieWang0528

bqcs, 2024-01-12:
> Thank you for your reply. I know where the problem lies

Can you please let me know what the issue is?

Petros626, 2024-10-04:
@bqcs Kitti uses training and test for the benchmark, OpenPCDet uses training and validation I guess.