open-mmlab/OpenPCDet #1424: How to use openpcdet to visualize both ground-true and predicted values simultaneously

Bryceder, 2023-08-07:
May I ask how to achieve simultaneous visualization of model predicted-values and ground-trues by modifying the demo.py or test.py files under Tools?  thanks！

jihanyang, 2023-08-07:
https://github.com/open-mmlab/OpenPCDet/blob/557d463793f6ff63f0301e851a047d29924fb1fe/tools/demo.py#L100. Check this function

Bryceder, 2023-08-07:
May I ask how this function should get its gt_boxes, taking the Kitti dataset as an example, should come from Kitti_dataset. py? This function seems to only run the script during the data preparation phase, so how should I obtain it?

closed 2023-08-17 by Bryceder

chenquansong, 2023-09-01:
Has this issue been resolved?

xback461007, 2024-01-16:
Has this issue been resolved?

Petros626, 2024-10-24:
@Bryceder @chenquansong @xback461007 Just catch your gt_boxes in 3D from the .pkl files. You can insert every gt in the loaded point cloud scene and apply after that inference of the model.