open-mmlab/OpenPCDet #1426: python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml ERROR

neverstoplearn, 2023-08-08:
when I follow the guidence to run python -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
I got this error:
Traceback (most recent call last):
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/runpy.py", line 188, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/runpy.py", line 111, in _get_module_details
    __import__(pkg_name)
  File "/xxx/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/xxx/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/xxx/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/site-packages/torch/jit/_script.py", line 1310, in script
    fn = torch._C._jit_script_compile(
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/site-packages/torch/jit/_recursive.py", line 838, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/site-packages/torch/jit/_script.py", line 1310, in script
    fn = torch._C._jit_script_compile(
RuntimeError: 
cannot statically infer the expected size of a list in this context:
  File "/xxx/miniconda3/envs/torch10/lib/python3.9/site-packages/kornia/geometry/conversions.py", line 553

    # this slightly awkward construction of the output shape is to satisfy torchscript
    output_shape = [*list(quaternion.shape[:-1]), 3, 3]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    matrix = matrix_flat.reshape(output_shape)
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/xxx/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )

how can I fix it? look forward your hlep.

Harry710887048, 2023-08-11:
It seems to be a problem with the kornia version, I replaced the version of it, and it works.

stu-zhangxl, 2023-08-12:
> It seems to be a problem with the kornia version, I replaced the version of it, and it works.

May I ask if you are using the Kornia version number? I am currently using version 0.7.0 and an error has been reported. Thank you

Harry710887048, 2023-08-12:
This might help you solve the problem.
Please try to reinstall kornia with `pip install kornia==0.5.8`.

Alga53, 2023-09-03:
> 

Thank you so much for sharing this! It solved my problem that haunts for so long. Love you bro~

closed 2023-10-19 by github-actions[bot]

Bennyliiiiii, 2024-02-26:
> This might help you solve the problem. Please try to reinstall kornia with `pip install kornia==0.5.8`.

great!

miaoxiannv, 2025-03-27:
> This might help you solve the problem. Please try to reinstall kornia with .`pip install kornia==0.5.8`

You are my God.！！！！！！

liyufan, 2025-04-23:
`kornia<0.7` is enough. See https://github.com/kornia/kornia/issues/2599
Also, has anyone tried #1722 ?