open-mmlab/OpenPCDet #1429: How can the model submitted on the KITTI official website be trained to achieve good results?（KITTI testing dataset）

find-star, 2023-08-10:
Hello. May I ask, how do I train my model when submitting it to the KITTI official website? Train 80 epochs with 80% data? What is the specific setting for the 'use slightly lower score threshold'? Thank you.

zbaishancha, 2023-09-07:
@find-star hello， May I ask if you know what to do?

sgrrlll, 2023-10-17:
> Hello. May I ask, how do I train my model when submitting it to the KITTI official website? Train 80 epochs with 80% data? What is the specific setting for the 'use slightly lower score threshold'? Thank you.

Do you know how to do it? Can you tell me?

sfy-1739, 2023-10-19:
Have you solved this problem? Please tell me

y1216942674, 2023-11-19:
What do I need to do to generate the file to submit kitti list

faziii0, 2023-12-15:
@y1216942674 have you figure it out

closed 2024-01-30 by github-actions[bot]