open-mmlab/OpenPCDet #1435: RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 225 but got size 226 for tensor number 1 in the list.

Li018, 2023-08-20:
python demo.py --cfg_file cfgs/kitti_models/pv_rcnn.yaml \
>     --ckpt pv_rcnn_8369.pth \
> --data_path 000005.bin
2023-08-20 12:03:41,698   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2023-08-20 12:03:41,699   INFO  Total number of samples:        1
/home/li/miniconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-08-20 12:03:48,219   INFO  ==> Loading parameters from checkpoint pv_rcnn_8369.pth to CPU
2023-08-20 12:03:48,351   INFO  ==> Done (loaded 367/367)
2023-08-20 12:03:48,848   INFO  Visualized sample index:        1
Traceback (most recent call last):
  File "demo.py", line 116, in <module>
    main()
  File "demo.py", line 102, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/li/OpenPCDet/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/li/miniconda3/envs/OpenPCDet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1102, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/li/OpenPCDet/pcdet/models/backbones_2d/base_bev_backbone.py", line 103, in forward
    x = torch.cat(ups, dim=1)
RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 225 but got size 226 for tensor number 1 in the list.
why I cant Visualize it ？  is some problem in dataset?

Li018, 2023-08-20:
pointpillar_7728.pth  can be used , but second and PV-RCNN demo  will makes RuntimeError:

closed 2023-10-04 by github-actions[bot]

Chenhongchang, 2024-04-27:
@Li018 hello! how can you fix this problem

Li018, 2024-05-29:
> @Li018 hello! how can you fix this problem

Sorry, I forgot, but I remember it was based on the following
https://www.cnblogs.com/kin-zhang/p/17002980.html