open-mmlab/OpenPCDet #1439: Use xyz to train

BarryLoveBerry, 2023-08-23:
Hi all,
I am trying to use xyz as feature only to train the model on the OpenPCDet framwork. But it doesn't work, is there anyone has the related experience about it?

tyutlk, 2023-08-24:
Yes,I've tried so. And some parameters (4 dimensions)for xyz & intensity are hard coded in PV-RCNN model ,so just concate an empty column after xyz will work.

jihanyang, 2023-08-24:
Actually, this should be very easy to implement. Take KITTI as an example, you just need to modify the used_feature_list to ['x', 'y', 'z']
https://github.com/open-mmlab/OpenPCDet/blob/bce886d6e36e3deaec2ce6f8b54b133fcadf62d6/tools/cfgs/dataset_configs/kitti_dataset.yaml#L49-L50

BarryLoveBerry, 2023-09-06:
> 

thanks, this issue has been solved.

closed 2023-09-06 by BarryLoveBerry