open-mmlab/OpenPCDet #1441: Multimodal .yaml

Michal12Zurawka, 2023-08-27:
Hello, everyone, want to ask about openpcdet using multimodal in waymo od kitti dataset. Actually i work on my master degree, my topic is about data fusion so i need to use multimodal. Have you any idea or already created .yaml to use multimodal on different dataset then nuscenes?

BBBeatrice, 2023-08-27:
@Michal12Zurawka ,hi , I am now using a model based on openPCDet for multimodal research, the model which I am referering to is https://github.com/hailanyi/VirConv, there is .yaml in this repository, maybe you can get some advice from this code. BTW, do you also use argoverse dataset as input? I want change the input to argoverse dataset, but I have no idea how to change .yaml to fit the new dataset.

chenshi3, 2023-08-27:
OpenPCDet now offers multi-modal support for the Nuscenes dataset. If you intend to utilize multimodal data with the Waymo dataset, you'll need to  reconfigure the [dataloader](https://github.com/open-mmlab/OpenPCDet/blob/bce886d6e36e3deaec2ce6f8b54b133fcadf62d6/pcdet/datasets/waymo/waymo_dataset.py#L23) and [utils](https://github.com/open-mmlab/OpenPCDet/blob/master/pcdet/datasets/waymo/waymo_utils.py).

Michal12Zurawka, 2023-08-29:
Generally i want change structure of dataset to KITTI structure to easly compare dependency of dataset to

Michal12Zurawka, 2023-10-11:
unfortunetly i give up with this, actually i fighting with https://github.com/anshulpaigwar/Frustum-Pointpillars

closed 2023-11-25 by github-actions[bot]