open-mmlab/OpenPCDet #1442: More effective implementation of farthest point sampling

hanm2019, 2023-08-28:
I have implemented a more effective version of farthest point sampling in both CUDA and CPP. The test results show that the CUDA version achieves a 3-4x speedup against the current implementation without any accuracy loss. It will help the farthest point sampling kernel applied in large-scale point clouds. 

However, I failed to integrate the code into the Python API. Maybe some volunteers could finish it.

1. [the CPU implementation of FPS](https://github.com/hanm2019/FPS_CPU)
2. [the GPU implementation of FPS](https://github.com/hanm2019/FPS_GPU)

NLCharles, 2023-08-29:
I suppose wrapping your kernel  into pytorch function takes trival efforts. Did you implement the backward functions?

hanm2019, 2023-08-29:
No, I only implement the forward functions. I think the backward functions can be reused because my kernel returns the same sampling result as the current implementation.

closed 2023-10-13 by github-actions[bot]