open-mmlab/OpenPCDet #1446: PointPillar: High AP but 0 recall_roi

yjwong1999, 2023-09-04:
Hi OpenPCDet teams,

**Background**
First and foremost, this is a fantastic repo! Absolutely appreciate the team's effort!
I'm implementing my custom dataset with pure point cloud.
For the use case, I don't mind to use IoU threshold of 0.3.
So the recall_roi and recall_rcnn for other IoU threshold is not a concern for me.

**Results**
My AP scores are high (as high as 95-97%).
However, my recall_roi_0.3 is only 0.00, while recall_rcnn_0.3 is 0.86.
I have attachedmy score at the end of the thread.

**My questions**
1. why is the recall_roi only 0, while recall_rcnn can reach higher value of 0.86?
2. if I am satisfied with the prediction after inspecting the demo + high AP, can i ignore the low recall_roi?

@jihanyang Can you assist me in this matter? thanks a lot in advance!

**Recall**
2023-09-04 11:23:22,387   INFO  recall_roi_0.3: 0.000000
2023-09-04 11:23:22,387   INFO  recall_rcnn_0.3: 0.865323
2023-09-04 11:23:22,387   INFO  recall_roi_0.5: 0.000000
2023-09-04 11:23:22,387   INFO  recall_rcnn_0.5: 0.706452
2023-09-04 11:23:22,387   INFO  recall_roi_0.7: 0.000000
2023-09-04 11:23:22,387   INFO  recall_rcnn_0.7: 0.376613

**AP (car)**
Car AP@0.70, 0.70, 0.70:
bbox AP:97.8221, 97.8221, 97.8221
bev  AP:26.2243, 26.2243, 26.2243
3d   AP:16.6734, 16.6734, 16.6734
aos  AP:60.44, 60.44, 60.44
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:98.9003, 98.9003, 98.9003
bev  AP:25.0477, 25.0477, 25.0477
3d   AP:14.8812, 14.8812, 14.8812
aos  AP:61.11, 61.11, 61.11
Car AP@0.70, 0.50, 0.50:
bbox AP:97.8221, 97.8221, 97.8221
bev  AP:51.4914, 51.4914, 51.4914
3d   AP:48.3760, 48.3760, 48.3760
aos  AP:60.44, 60.44, 60.44
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:98.9003, 98.9003, 98.9003
bev  AP:54.1587, 54.1587, 54.1587
3d   AP:47.8132, 47.8132, 47.8132
aos  AP:61.11, 61.11, 61.11

**AP (Pedestrian)**
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:94.3198, 94.3198, 94.3198
bev  AP:67.6892, 67.6892, 67.6892
3d   AP:57.7386, 57.7386, 57.7386
aos  AP:72.17, 72.17, 72.17
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:95.7393, 95.7393, 95.7393
bev  AP:68.4247, 68.4247, 68.4247
3d   AP:59.2235, 59.2235, 59.2235
aos  AP:71.90, 71.90, 71.90
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:94.3198, 94.3198, 94.3198
bev  AP:79.0836, 79.0836, 79.0836
3d   AP:76.8043, 76.8043, 76.8043
aos  AP:72.17, 72.17, 72.17
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:95.7393, 95.7393, 95.7393
bev  AP:79.7241, 79.7241, 79.7241
3d   AP:77.1870, 77.1870, 77.1870
aos  AP:71.90, 71.90, 71.90

lijin123red, 2023-09-21:
hello,i have the same question,did you resolve this  question?Looking forward to your reply!

jihanyang, 2023-09-21:
RoI recall only valid for detectors has ROI head. However, SECOND and PointPillar do not have ROI Head.

closed 2023-09-21 by jihanyang

lijin123red, 2023-09-21:
> RoI recall only valid for detectors has ROI head. However, SECOND and PointPillar do not have ROI Head.

Thank you very much for your reply!!!!!!!!!!