open-mmlab/OpenPCDet #1453: Are the traininig params saved in the tools/cfgs/kitti_models/pointpillar.yaml file used for the pretrained model provided?

zppppppx, 2023-09-08:
I am a little confused about the number of epochs. When I use the pre-trained model to do the evaluation, the logs shows that this is the evaluation of epoch-7728
```
...

2023-08-23 01:35:46,343   INFO  Loading KITTI dataset
2023-08-23 01:35:46,904   INFO  Total samples for KITTI dataset: 7538
/usr/local/lib/python3.8/dist-packages/torch/functional.py:478: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  ../aten/src/ATen/native/TensorShape.cpp:2894.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-08-23 01:35:48,413   INFO  ==> Loading parameters from checkpoint ./checkpoints/pointpillar_7728.pth to GPU
2023-08-23 01:35:48,970   INFO  ==> Done (loaded 127/127)
2023-08-23 01:35:48,976   INFO  *************** EPOCH 7728 EVALUATION *****************
...
```

But when I load the checkpoint, it seems it is epoch 80 (the same as that in the yaml file), and I cannot start training the model if I use the flag `--ckpt` to load the file and assign a training epoch smaller than 80. Why's that?

zppppppx, 2023-09-08:
Basically I want to know how you obtain the pretrained model from the kitti dataset using pointpillars. I had tried training it myself with 80 epochs but the validation accuracy is much worse than the pretrained model.

yuwenmichael, 2023-09-12:
Yes, I did the training from scratch and it's totally fine (compared to the original paper). However, I changed some parameters in the `pointpillar_train.yaml` file to better align with the paper.
Here is what I changed:
```
MAX_NUMBER_OF_VOXELS: {
            'train': 12000,
            'test': 40000
}
....
class_name': 'Car',
'anchor_sizes': [[3.9, 1.6, 1.5]],
```
Don't know if this actually helps tho hahaha.

Here is the final result:
```
Car AP@0.70, 0.70, 0.70:
bbox AP:90.7830, 89.6159, 88.6891
bev  AP:89.6850, 87.1631, 83.8572
3d   AP:86.3287, 76.9424, 74.9217
aos  AP:90.77, 89.43, 88.37
Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:65.8095, 61.9866, 59.0696
bev  AP:60.5871, 54.8542, 51.6975
3d   AP:53.9660, 48.6444, 45.8279
aos  AP:45.83, 43.22, 41.03
Cyclist AP@0.50, 0.50, 0.50:
bbox AP:89.3821, 74.4809, 70.8870
bev  AP:83.8311, 68.2058, 63.3370
3d   AP:79.1489, 61.4103, 58.6771
aos  AP:88.20, 72.37, 68.68
```

zppppppx, 2023-09-12:
> Yes, I did the training from scratch and it's totally fine (compared to the original paper). However, I changed some parameters in the `pointpillar_train.yaml` file to better align with the paper. [...]

OK thanks, I am not sure where it went wrong previously, but I managed to achieve a similar precision. I noticed that you also have a lower precision for Pedestrian Class compared to the provided pretrained model. I only changed the batch size so that it could be closer to their experimenting environment that `batch_size = 32` for 8 GPUs. Maybe that is one factor that could make a difference. May I know how many GPUs you were using for training?

yuwenmichael, 2023-09-14:
I trained on 8 GPUs with default `batch size = 4`. I think the reason for the low AP in the Pedestrian Class is that the MAX_NUMBER_OF_VOXELS for training is only 12000, which may not be sufficient to detect small objects, like pedestrians in this case. Probably setting it back to 16000 will have a similar performance.

zppppppx, 2023-10-07:
That could be one reason. Thanks for your help anyway, though I still cannot regenerate the results lol. By the way, do you mind offering your package list and anything like the version of the cuda? I noticed that our project forked OpenPCDet about one year ago, and I want to try the current version but there is always a conflict. Thanks again!

closed 2023-11-21 by github-actions[bot]