open-mmlab/OpenPCDet #1454: When running python setup.py develop, I encountered an error. What could be the issue?

sgrrlll, 2023-09-11:
I'm running it under an environment copied from another computer, and CUDA is already installed, and the nvcc -V command does not report any errors.Error message as follows：
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/7] :/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_attention_indices_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/build_attention_indices_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_attention_indices_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=votr_ops_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_attention_indices_gpu.o
:/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_attention_indices_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/build_attention_indices_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_attention_indices_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=votr_ops_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
/bin/sh: 1: :/home/shuxue3/.local/cuda-12.1/bin/nvcc: not found
[2/7] :/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_mapping_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/build_mapping_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_mapping_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=votr_ops_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_mapping_gpu.o
:/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_mapping_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/build_mapping_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/build_mapping_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=votr_ops_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
/bin/sh: 1: :/home/shuxue3/.local/cuda-12.1/bin/nvcc: not found
[3/7] :/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/group_features_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/group_features_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/group_features_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=votr_ops_cuda -D_GLIBCXX_USE_CXX11_ABI=0 -gencode=arch=compute_86,code=compute_86 -gencode=arch=compute_86,code=sm_86 -std=c++14
FAILED: /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/group_features_gpu.o
:/home/shuxue3/.local/cuda-12.1/bin/nvcc --generate-dependencies-with-compile --dependency-output /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/group_features_gpu.o.d -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/TH -I/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/include/THC -I:/home/shuxue3/.local/cuda-12.1/include -I/home/shuxue3/anaconda3/envs/casa/include/python3.8 -c -c /home/shuxue3/lyk/CaSA-3DHANet/pcdet/ops/votr_ops/src/group_features_gpu.cu -o /home/shuxue3/lyk/CaSA-3DHANet/build/temp.linux-x86_64-cpython-38/pcdet/ops/votr_ops/src/group_features_gpu.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DP [... cut ...]
  File "/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 529, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1354, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/shuxue3/anaconda3/envs/casa/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1683, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension

SinaDavari, 2023-10-04:
Same error here... Any solutions?

kbratsy, 2023-10-13:
I also encounter the same error while compiling. I would appreciate it if you could provide any guidance.

sgrrlll, 2023-10-16:
> Same error here... Any solutions?

The CUDA version is not right, I lowered the version and it was fine

sgrrlll, 2023-10-16:
> I also encounter the same error while compiling. I would appreciate it if you could provide any guidance.

The CUDA version is not right, I lowered the version and it was fine