open-mmlab/OpenPCDet #1463: Question on KITTI test set vs val set

NiranjanRavi1993, 2023-09-24:
Hi all,
        Thank you for the excellent repo. I have a question regarding evaluation in KITTI test set. I am currently working with proposing a new approach in point-pillars. To do this, I started with a baseline model (developed with OpenPCDet) which is already evaluated in KITTI test set. I re-trained the network with my proposed changes and baseline network on "train" set and tested both networks on "val" set. My proposed model indicates better performance. 
So, as a next step, I trained the network with my proposed changes on "train+val" dataset. When I evaluate on KITTI test set, its lower compared to baseline model. 
Since, we have to submit the best checkpoint files to KITTI, I cant test it multiple times as well. Did anyone face a similar issue? Is there other ways I can debug this issue more? I am unable to understand this correlation. 
Any inputs would be helpful. 
Thanks in advance.

closed 2023-11-08 by github-actions[bot]

bqcs, 2023-12-29:
> Hi all, Thank you for the excellent repo. I have a question regarding evaluation in KITTI test set. I am currently working with proposing a new approach in point-pillars. To do this, I started with a baseline model (developed with OpenPCDet) which is already evaluated in KITTI test set. I re-trained the network with my proposed changes and baseline network on "train" set and tested both networks on "val" set. My proposed model indicates better performance. So, as a next step, I trained the network with my proposed changes on "train+val" dataset. When I evaluate on KITTI test set, its lower compared to baseline model. Since, we have to submit the best checkpoint files to KITTI, I cant test it multiple times as well. Did anyone face a similar issue? Any way I can evaluate my results locally before uploading to KITTI server? I am unable to understand this correlation. Any inputs would be helpful. Thanks in advance.

hi,Can you tell me how to evaluate the test set and get results on the official KITTI? What is the process like?

NiranjanRavi1993, 2024-01-02:
Sure, its simple. Change the dataset from val to test and change the flag store_true to True in test.py. Check OpenPCDet repo issues for further information.

bqcs, 2024-01-12:
> Thank you for the excellent repo. I have a question regarding evaluation in KITTI test set. I am currently working with proposing a new approach in point-pillars. To do this, I started with a [...]

Thank you for your reply, it's very helpful! I also submitted to the KITTI official and checked the results. I encountered a situation similar to yours. In the test set for Pedestrian (3D Detection), I got 40.74% for Easy, 33.57% for Moderate, and 31.32% for Hard. In the validation set, I obtained 57.79%, 52.45%, and 49.20% for Easy, Moderate, and Hard, respectively. I understand that the performance on the test set may be slightly lower than that on the validation set, but the difference in my results seems unusually large. Have you tried using the official pointpillar_7728.pth provided by KITTI for detection and then submitting the results to compare?

NiranjanRavi1993, 2024-01-12:
Hi, I did not find an answer why this issue was happening. But when I researched into few articles in this domain, there was one observation. The networks were not trained in complete train+val dataset but at some percentage of val is used along with train. I am not sure what percentage of it.

bqcs, 2024-01-12:
> Hi, I did not find an answer why this issue was happening. But when I researched into few articles in this domain, there was one observation. The networks were not trained in complete train+val dataset but at some percentage of val is used along with train. I am not sure what percentage of it.

[image]
"PV-RCNN" paper mentions that the results on the test set are achieved using a customized 80% of the training set. I believe it would be interesting to compare submissions using both 80% and 100% of the training set. I found the original author's response in the following comments.      I have another question， do you know which parameters or files need to be modified to customize the selection of the number of training set samples?

NiranjanRavi1993, 2024-01-12:
I think this is one of the articles I too have seen. Unfortunately, no. Since, I had only three attempts for a month to test in Kitti , I couldnt explore further.

NiranjanRavi1993, 2024-01-12:
Please comment back here, if you manage to get the results by the split method.

bqcs, 2024-01-14:
@NiranjanRavi1993 。Hi，when I split the training and validation sets in an 80-20 ratio, the performance on the test set is better with a 50-50 split. 
Additionally, I have a crucial issue. I used the official pre-trained model from the PCDet repository, trained it with a 50-50 split, and submitted it for testing. The results for pedestrian classes are: 40.65%, 32.76%, 30.20%.
 However, the PointPillars official test results are 52.08%, 43.53%, 41.49%. Do you know the reason for this? I noticed in your mention  #1116  that the pre-trained models/official report results in the PCDet software repository dropped by 1-3 , but mine dropped by 10-12。

bqcs, 2024-01-14:
@NiranjanRavi1993 ，Strangely, in terms of performance on the test set, the 'car' category shows improvement compared to the original paper, but the categories of 'pedestrian' and 'cyclist' exhibit a significant decrease.

NiranjanRavi1993, 2024-01-14:
@bqcs  Yes, I agree with your inputs. I have seen a performance drop of 1-5% but not more. I have actually observed the same problem in other networks as well. That is, take a pre-trained model and with no additional training, I tried to observe the results in KITTI test set and it was still lower than the reported results at time. I did not find a reason. But I remember seeing one of the comments stating the test data set is not stable and val data is sometimes utilized to observe the performance difference (This works, when you have some network changes and want to observe performance).

GP-Bone, 2024-01-15:
> @NiranjanRavi1993 ，Strangely, in terms of performance on the test set, the 'car' category shows improvement compared to the original paper, but the categories of 'pedestrian' and 'cyclist' exhibit a significant decrease.

I found that the best checkpoint for 'car',' pedestrian' and 'cyclist' is not the same checkpoint. Maybe the three class objects in the test set are a splice of the results of the three checkpoints.

GP-Bone, 2024-01-15:
> @NiranjanRavi1993 。Hi，when I split the training and validation sets in an 80-20 ratio, the performance on the test set is better with a 50-50 split. Additionally, I have a crucial issue. I used the [...]

Hi, it seems you have solved the performance problem of Car in Kitti test set, can you describe how to divide the whole dataset in 8:2 ratio. Do you use random allocation? Looking forward to your reply as this has really been bothering me for a long time.

bqcs, 2024-01-15:
> Hi, it seems you have solved the performance problem of Car in Kitti test set, can you describe how to divide the whole dataset in 8:2 ratio. Do you use random allocation? Looking forward to your [...]

Sure,I  added a certain number of randomly selected files from the "val.txt" file to the "train.txt" file, in order to achieve an 80:20 ratio from the original 50:50 ratio.
Additionally, I would like to ask if you know how to modify the configuration file(like pointpillar.yaml) to detect a single class/category?

GP-Bone, 2024-01-15:
> Sure,I added a certain number of randomly selected files from the "val.txt" file to the "train.txt" file, in order to achieve an 80:20 ratio from the original 50:50 ratio. Additionally, I would like [...]

Thanks for your reply. If you want to inference or train single class , just delete some information about 3 classes. Such as pointpillars as follows:

CLASS_NAMES: ['Car'] #['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG: 
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
    POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
    DATA_PROCESSOR:
        - NAME: mask_points_and_boxes_outside_range
          REMOVE_OUTSIDE_BOXES: True

        - NAME: shuffle_points
          SHUFFLE_ENABLED: {
            'train': True,
            'test': False
          }

        - NAME: transform_points_to_voxels
          VOXEL_SIZE: [0.16, 0.16, 4]
          MAX_POINTS_PER_VOXEL: 32
          MAX_NUMBER_OF_VOXELS: {
            'train': 16000,
            'test': 40000
          }
    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: gt_sampling
              USE_ROAD_PLANE: True
              DB_INFO_PATH:
                  - kitti_dbinfos_train.pkl
              PREPARE: {
                 filter_by_min_points: ['Car:5'],   #['Car:5', 'Pedestrian:5', 'Cyclist:5']
                 filter_by_difficulty: [-1],
              }

              SAMPLE_GROUPS: ['Car:15']  #['Car:15','Pedestrian:15', 'Cyclist:15']
              NUM_POINT_FEATURES: 4
              DATABASE_WITH_FAKELIDAR: False
              REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
              LIMIT_WHOLE_SCENE: False

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: PointPillar

    VFE:
        NAME: PillarVFE
        WITH_DISTANCE: False
        USE_ABSLOTE_XYZ: True
        USE_NORM: True
        NUM_FILTERS: [64]

    MAP_TO_BEV:
        NAME: PointPillarScatter
        NUM_BEV_FEATURES: 64

    BACKBONE_2D:
        NAME: BaseBEVBackbone
        LAYER_NUMS: [3, 5, 5]
        LAYER_STRIDES: [2, 2, 2]
        NUM_FILTERS: [64, 128, 256]
        UPSAMPLE_STRIDES: [1, 2, 4]
        NUM_UPSAMPLE_FILTERS: [128, 128, 128]

    DENSE_HEAD:
        NAME: AnchorHeadSingle
        CLASS_AGNOSTIC: False

        USE_DIRECTION_CLASSIFIER: True
        DIR_OFFSET: 0.78539
        DIR_LIMIT_OFFSET: 0.0
        NUM_DIR_BINS: 2

        ANCHOR_GENERATOR_CONFIG: [
            {
                'class_name': 'Car',
                'anchor_sizes': [[3.9, 1.6, 1.56]],
                'anchor_rotations': [0, 1.57],
                'anchor_bottom_heights': [-1.78],
                'align_center': False,
                'feature_map_stride': 2,
                'matched_threshold': 0.6,
                'unmatched_threshold': 0.45
            },
#            {
#                'class_name': 'Pedestrian',
#                'anchor_sizes': [[0.8, 0.6, 1.73]],
#                'anchor_rotations': [0, 1.57],
#                'anchor_bottom_heights': [-0.6],
#                'align_center': False,
#                'feature_map_stride': 2,
#                'matched_threshold': 0.5,
#                'unmatched_threshold': 0.35
#            },
#            {
#                'class_name': 'Cyclist',
#                'anchor_sizes': [[1.76, 0.6, 1.73]],
#                'anchor_rotations': [0, 1.57],
#                'anchor_bottom_heights': [-0.6],
#                'align_center': False,
#                'feature_map_stride': 2,
#                'matched_threshold': 0.5,
#                'unmatched_threshold': 0.35
#            }
        ]

        TARGET_ASSIGNER_CONFIG:
            NAME: AxisAlignedTargetAssigner
            POS_FRACTION: -1.0
            SAMPLE_SIZE: 512
            NORM_BY_NUM_EXAMPLES: False
            MATCH_HEIGHT: False
            BOX_CODER: ResidualCoder

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 2.0,
                'dir_weight': 0.2,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 1.0]
            }

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]
        SCORE_THRESH: 0.1
        OUTPUT_RAW_SCORE: False

        EVAL_METRIC: kitti

        NMS_CONFIG:
            MULTI_CLASSES_NMS: False
            NMS_TYPE: nms_gpu
            NMS_THRESH: 0.01
            NMS_PRE_MAXSIZE: 4096
            NMS_POST_MAXSIZE: 500

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 80

    OPTIMIZER: adam_onecycle
    LR: 0.003
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10

bqcs, 2024-01-15:
> Thanks for your reply. If you want to inference or train single class , just delete some information about 3 classes. Such as pointpillars as follows:

I understand now, thank you for your response!

GP-Bone, 2024-01-15:
> I understand now, thank you for your response!

Thanks for your suggestions as well. I was hoping that the 8:2 distribution would achieve the mAP of the paper. If this works, I'll return to the comments. All the best to you.

GP-Bone, 2024-02-27:
> > Sure, its simple. Change the dataset from val to test and change the flag store_true to True in test.py. Check OpenPCDet repo issues for further information.
> 
> hi bro. I also encountered this problem. When I wanted to test on the kitti data set, I didn’t know where my modifications were. The program on the terminal never stopped, resulting in no results in the end. How is this going? Thank you very much for your answer
> 
> (openpcdet1) lrt@lrt:~/OpenPCDet/tools$ python test.py --cfg_file cfgs/kitti_models/second.yaml --batch_size 4 --ckpt pth/second_7862.pth 2024-02-27 15:03:54,706 INFO **Start logging** 2024-02-27 15:03:54,706 INFO CUDA_VISIBLE_DEVICES=ALL 2024-02-27 15:03:54,706 INFO cfg_file cfgs/kitti_models/second.yaml 2024-02-27 15:03:54,706 INFO batch_size 4 2024-02-27 15:03:54,706 INFO workers 4 2024-02-27 15:03:54,706 INFO extra_tag default 2024-02-27 15:03:54,707 INFO ckpt pth/second_7862.pth 2024-02-27 15:03:54,707 INFO pretrained_model None 2024-02-27 15:03:54,707 INFO launcher none 2024-02-27 15:03:54,707 INFO tcp_port 18888 2024-02-27 15:03:54,707 INFO local_rank 0 2024-02-27 15:03:54,707 INFO set_cfgs None 2024-02-27 15:03:54,707 INFO max_waiting_mins 30 2024-02-27 15:03:54,707 INFO start_epoch 0 2024-02-27 15:03:54,707 INFO eval_tag default 2024-02-27 15:03:54,707 INFO eval_all True 2024-02-27 15:03:54,707 INFO ckpt_dir None 2024-02-27 15:03:54,707 INFO save_to_file True 2024-02-27 15:03:54,707 INFO infer_time True 2024-02-27 15:03:54,707 INFO cfg.ROOT_DIR: /home/lrt/OpenPCDet 2024-02-27 15:03:54,707 INFO cfg.LOCAL_RANK: 0 2024-02-27 15:03:54,707 INFO cfg.CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist'] 2024-02-27 15:03:54,707 INFO ----------- DATA_CONFIG ----------- 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATASET: KittiDataset 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_PATH: ../data/kitti 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [0, -40, -3, 70.4, 40, 1] 2024-02-27 15:03:54,707 INFO ----------- DATA_SPLIT ----------- 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_SPLIT.train: train 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_SPLIT.test: test 2024-02-27 15:03:54,707 INFO ----------- INFO_PATH ----------- 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.INFO_PATH.train: ['kitti_infos_train.pkl'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.INFO_PATH.test: ['kitti_infos_test.pkl'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.GET_ITEM_LIST: ['points'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.FOV_POINTS_ONLY: True 2024-02-27 15:03:54,707 INFO ----------- DATA_AUGMENTOR ----------- 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'USE_ROAD_PLANE': False, 'DB_INFO_PATH': ['kitti_dbinfos_train.pkl'], 'PREPARE': {'filter_by_min_points': ['Car:5', 'Pedestrian:5', 'Cyclist:5'], 'filter_by_difficulty': [-1]}, 'SAMPLE_GROUPS': ['Car:20', 'Pedestrian:15', 'Cyclist:15'], 'NUM_POINT_FEATURES': 4, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}] 2024-02-27 15:03:54,707 INFO ----------- POINT_FEATURE_ENCODING ----------- 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity'] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': False}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.05, 0.05, 0.1], 'MAX_POINTS_PER_VOXEL': 5, 'MAX_NUMBER_OF_VOXELS': {'train': 16000, 'test': 40000}}] 2024-02-27 15:03:54,707 INFO cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml 2024-02-27 15:03:54,707 INFO ----------- MODEL ----------- 2024-02-27 15:03:54,707 INFO cfg.MODEL.NAME: SECONDNet 2024-02-27 15:03:54,707 INFO ----------- VFE ----------- 2024-02-27 15:03:54,707 INFO cfg.MODEL.VFE.NAME: MeanVFE 2024-02-27 15:03:54,707 INFO ----------- BACKBONE_3D ----------- 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_3D.NAME: VoxelBackBone8x 2024-02-27 15:03:54,708 INFO ----------- MAP_TO_BEV ----------- 2024-02-27 15:03:54,708 INFO cfg.MODEL.MAP_TO_BEV.NAME: HeightCompression 2024-02-27 15:03:54,708 INFO cfg.MODEL.MAP_TO_BEV.NUM_BEV_FEATURES: 256 2024-02-27 15:03:54,708 INFO ----------- BACKBONE_2D ----------- 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.NAME: BaseBEVBackbone 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.LAYER_NUMS: [5, 5] 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.LAYER_STRIDES: [1, 2] 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.NUM_FILTERS: [128, 256] 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.UPSAMPLE_STRIDES: [1, 2] 2024-02-27 15:03:54,708 INFO cfg.MODEL.BACKBONE_2D.NUM_UPSAMPLE_FILTERS: [256, 256] 2024-02-27 15:03:54,708 INFO ----------- DENSE_HEAD ----------- 2024-02-27 15:03:54,708 INFO cfg.MODEL.DENSE_HEAD.NAME: AnchorHeadSingle 2024-02-27 15:03:54,7 [... cut ...]
        'train': train,
        'test': test         # val or test
    }
    INFO_PATH: {
        'train': [ kitti_infos_train.pkl ],               # [ kitti_infos_train.pkl ] or [ kitti_infos_trainval.pkl ]
        'test': [ kitti_infos_test.pkl ],                 # [ kitti_infos_val.pkl ] or [ kitti_infos_test.pkl ]
    }

‘’’

And you can run this code to test at the test set:
'''
python test.py --workers 4 --cfg_file cfgs/kitti_models/second.yaml --batch_size 1 \
     --ckpt pth/second_7862.pth --save_to_file
'''

Karkers, 2024-04-20:
Hello, I would like to know if all the training data here refers to a training set divided by an 8:2 ratio, or if all the images are used for training without dividing the validation set. If that's the case, I would like to know how to proceed. Thank you!

Petros626, 2024-10-24:
@NiranjanRavi1993 Hey, can you tell me when using `test.py` will the the kitti_infos_val.pkl be loaded and the data pre-processed through DataProcessor and PointFeatureEncoder?

Petros626, 2024-10-24:
> Hello, I would like to know if all the training data here refers to a training set divided by an 8:2 ratio, or if all the images are used for training without dividing the validation set. If that's the case, I would like to know how to proceed. Thank you!

The standard splitting when using KITTI dataset is ~50:50 that' what i can observe from the .txt files in ImageSets folder.