open-mmlab/OpenPCDet #1469: about --infer_time.   How to get inference time?

strugglingguoguo, 2023-10-01:
I found this sentence in the train log file:   Generate lable finished(sec_per_example:0.0261 second).Does this mean inference time?I really hope you can help me answer this question,thank you.

yjwong1999, 2023-10-16:
Hi there,
I have the same doubts too. Have you find out the meaning of the ```Generate lable finished(sec_per_example:0.0261 second)```?

strugglingguoguo, 2023-10-17:
> 你好， 我也有同样的疑问。你找出其中的含义了吗`Generate lable finished(sec_per_example:0.0261 second)`？

Sorry, I don't have a clear answer either. But I presume this is inference time.

yjwong1999, 2023-10-29:
Hi @strugglingguoguo and anyone who have the same concerns

I have checked the codebase
Based on: [Line 94 to 95 of tools/eval_utils/eval_utils.py](https://github.com/open-mmlab/OpenPCDet/blob/255db8f02a8bd07211d2c91f54602d63c4c93356/tools/eval_utils/eval_utils.py#L94)

```
sec_per_example = (time.time() - start_time) / len(dataloader.dataset)
logger.info('Generate label finished(sec_per_example: %.4f second).' % sec_per_example)
```
You can see that sec_per_example is the total time taken / total data
Hence, it is indeed the inference speed
Note that the inference speed does not include the visualization, just the model.forward() and label generation

strugglingguoguo, 2023-10-30:
> Hi @strugglingguoguo and anyone who have the same concerns
>
> I have checked the codebase Based on: [Line 94 to 95 of [...]

I‘m so thankful for your reply.

closed 2023-12-15 by github-actions[bot]