open-mmlab/OpenPCDet #1470: Colab results

Agbeli, 2023-10-02:
Thank you guys for making this code available to the public. I running demo code and I always get error I would be glad if someone could help out     return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )

Qu0rise, 2023-10-05:
Hello @Agbeli,

Thank you for your engagement with OpenPCDet and for bringing this issue to light.

The error you're encountering is indeed related to a change in the `kornia` library. Specifically, `QuaternionCoeffOrder` has been deprecated and removed in versions 0.7.0 and above, as detailed in [kornia issue #2599](https://github.com/kornia/kornia/issues/2599). This alteration is likely the root cause of the error you're experiencing.

🛠 **Solution:**
You might consider downgrading `kornia` to a version that still includes `QuaternionCoeffOrder`. Version 0.6.1 is one such version that you might utilize:
\```bash
pip install kornia==0.6.1
\```

🔄 **Alternative Approach:**
Alternatively, if downgrading is not a preferred option, you might consider modifying the code to be compatible with the latest version of `kornia`. This would involve adjusting the function calls that utilize `QuaternionCoeffOrder` to align with the new API.

📝 **Note:**
Ensuring that your environment aligns with the project’s requirements is pivotal to avoid such issues. The `requirements.txt` or `environment.yml` file in the project repository might specify the versions of dependencies that the project has been tested with, providing a stable environment for execution.

yjwong1999, 2023-10-09:
Hi @Agbeli 

I faced the same problem when running the codes in Colab.
As mentioned by Qu0rise, you might need to consider downgrading ```kornia```
In my case:
```pip install kornia==0.5.8```

Agbeli, 2023-10-10:
Thank you guys

closed 2023-11-24 by github-actions[bot]