open-mmlab/OpenPCDet #1486: No module named 'pcdet.datasets.argo2.argo2_utils'

thiesymon, 2023-10-16:
I have run  python setup.py develop with successfule result:Finished processing dependencies for pcdet==0.6.0+255db8f
 when I run a demo in directory OpenPCDet/tools: python demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml  --ckpt pointpillar_7728.pth  --data_path ~/nets/OpenPCDet/data/kitti/training/velodyne/000008.bin
 I shows :

```
Traceback (most recent call last):
  File "demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/thiesy/miniconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/home/thiesy/miniconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
ModuleNotFoundError: No module named 'pcdet.datasets.argo2.argo2_utils'
```

 who can help with above ???

closed 2023-12-01 by github-actions[bot]

Zixiu99, 2024-05-17:
Have you solved the problem? I had the same problem.

Zixiu99, 2024-05-17:
I've had that problem. To resolve the issue, replace the pcdet folder in/home/thiesy/miniconda3/envs/openpcdet/lib/python3.8/site-packages/pcdet-0.6.0+255db8f-py3.8-linux-x86_64.egg/pcdet with the pcdet folder from the project.