open-mmlab/OpenPCDet #1487: Error when launching demo.py

IrumaMegumi, 2023-10-16:
When I run demo.py, an error occurred as follows:
Traceback (most recent call last):
  File "/media/iruma/Data/AIR/radar/OpenPCDet/test/demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/home/iruma/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/home/iruma/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/home/iruma/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1381, in script
    fn = torch._C._jit_script_compile(
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_recursive.py", line 1010, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/usr/local/lib/python3.10/dist-packages/torch/jit/_script.py", line 1381, in script
    fn = torch._C._jit_script_compile(
RuntimeError: 
cannot statically infer the expected size of a list in this context:
  File "/usr/local/lib/python3.10/dist-packages/kornia/geometry/conversions.py", line 553

    # this slightly awkward construction of the output shape is to satisfy torchscript
    output_shape = [*list(quaternion.shape[:-1]), 3, 3]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    matrix = matrix_flat.reshape(output_shape)
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/home/iruma/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )
It seems that there is something wrong in module kornia. How to solve it?

thiesymon, 2023-10-16:
change the kornia version down to 0.6.5 may works
`pip install kornia==0.6.5`

yjwong1999, 2023-10-16:
Hi @IrumaMegumi 
As mentioned by thiesymon, you need to downgrade your kornia version.
For my case, I need to downgrade to ```pip install kornia==0.5.8``` for it to work,
[Issue 1470: Colab results #1470](https://github.com/open-mmlab/OpenPCDet/issues/1470#issuecomment-1747974237)

thiesymon, 2023-10-16:
can you help to solve my isusse submited todayFrom Thiesy〽️

IrumaMegumi, 2023-10-16:
Thanks. Version 0.5.8 works!

closed 2023-12-01 by github-actions[bot]