open-mmlab/OpenPCDet #1488: AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

IrumaMegumi, 2023-10-16:
Traceback (most recent call last):
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train.py", line 230, in <module>
    main()
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train.py", line 117, in main
    train_set, train_loader, train_sampler = build_dataloader(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/__init__.py", line 57, in build_dataloader
    dataset = __all__[dataset_cfg.DATASET](
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/kitti/kitti_dataset.py", line 23, in __init__
    super().__init__(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/dataset.py", line 31, in __init__
    self.data_augmentor = DataAugmentor(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 24, in __init__
    cur_augmentor = getattr(self, cur_cfg.NAME)(config=cur_cfg)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/augmentor/data_augmentor.py", line 40, in gt_sampling
    db_sampler = database_sampler.DataBaseSampler(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/datasets/augmentor/database_sampler.py", line 35, in __init__
    sampler_cfg.DB_INFO_PATH[0] = sampler_cfg.BACKUP_DB_INFO['DB_INFO_PATH']
AttributeError: 'EasyDict' object has no attribute 'BACKUP_DB_INFO'

I have noticed my position of .pkl files, but the problem still exists. How to solve it?
My yaml files are as follows:
[image]

When I use Pycharm to debug, I found dict sampler_cfg does not have attribute 'BACKUP_INFO' as follows:
[image]

Besides, I want to know the time of training KITTI dataset using an RTX3090 or anRTX4090.Thanks.

IrumaMegumi, 2023-10-16:
I have solved it. Thanks.

closed 2023-10-16 by IrumaMegumi

3623687277, 2023-11-10:
您怎么解决的？能说一下吗？

AJAY31797, 2024-01-16:
I am getting the same error, how did you solve it @IrumaMegumi ?

Thanks.

Zixiu99, 2024-02-25:
> I have solved it. Thanks.

Can you share how you solved it? I'm having the same problem.

Zixiu99, 2024-02-25:
> I am getting the same error, how did you solve it @IrumaMegumi ?
> 
> Thanks.

May I ask if you've solved it now?

IrumaMegumi, 2024-03-13:
Sorry, I have benn busy recently. You should follow the following steps:
1. make a folder in data/kitti, called training.
2.go to the training folder you made just now and make 4 folders:calib,image_2,velodyne and label_2.
3.put your data into the four folders.
4.change your txt files in Imagesets folder, be the same to the file number in traning folders.
5.if you have installed OpenPCDet(means you have run the command:python3 setup.py develop),go to the folder of OpenPCDet, and run the command:
python3 -m pcdet.datasets.kitti.kitti_dataset create_kitti_infos tools/cfgs/dataset_configs/kitti_dataset.yaml
Have a try and good luck to you!