open-mmlab/OpenPCDet #1489: RuntimeError: handle_0 INTERNAL ASSERT FAILED at "../c10/cuda/driver_api.cpp":15, please report a bug to PyTorch.

IrumaMegumi, 2023-10-16:
Traceback (most recent call last):
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train.py", line 230, in <module>
    main()
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train.py", line 175, in main
    train_model(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train_utils/train_utils.py", line 180, in train_model
    accumulated_iter = train_one_epoch(
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/train_utils/train_utils.py", line 56, in train_one_epoch
    loss, tb_dict, disp_dict = model_func(model, batch)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/models/__init__.py", line 44, in model_func
    ret_dict, tb_dict, disp_dict = model(batch_dict)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/models/roi_heads/pvrcnn_head.py", line 151, in forward
    pooled_features = self.roi_grid_pool(batch_dict)  # (BxN, 6x6x6, C)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/models/roi_heads/pvrcnn_head.py", line 97, in roi_grid_pool
    pooled_points, pooled_features = self.roi_grid_pool_layer(
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/media/iruma/Data/AIR/radar/OpenPCDet/tools/../pcdet/ops/pointnet2/pointnet2_stack/pointnet2_modules.py", line 95, in forward
    new_features = self.mlps[k](new_features)  # (1, C, M1 + M2 ..., nsample)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/container.py", line 215, in forward
    input = module(input)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/modules/activation.py", line 101, in forward
    return F.relu(input, inplace=self.inplace)
  File "/usr/local/lib/python3.10/dist-packages/torch/nn/functional.py", line 1471, in relu
    result = torch.relu(input)
RuntimeError: handle_0 INTERNAL ASSERT FAILED at "../c10/cuda/driver_api.cpp":15, please report a bug to PyTorch.

When I use the train.py code, I found the error. I don't know how to solve it. Please help me. Thanks.

IrumaMegumi, 2023-10-16:
I have fixed it. Please be careful when you use CUDA12.

closed 2023-10-16 by IrumaMegumi

1356075561, 2023-10-17:
> I have fixed it. Please be careful when you use CUDA12.

I met the same error. Can you share how you fixed it please?

IrumaMegumi, 2023-10-17:
> > I have fixed it. Please be careful when you use CUDA12.
> 
> I met the same error. Can you share how you fixed it please?

I lowered my batch size to 1 and lowered my workers to 1. It worked well. When I use RTX4090, the error didn't happen. Personally I think it is like the error: cuda out of memory. Nothing to do with the code.

Yusjlalala, 2023-10-23:
It works when I lowered my batch size to 1, but how about other batch sizes?

IrumaMegumi, 2023-10-25:
> It works when I lowered my batch size to 1, but how about other batch sizes?

Pay attention to your num workers and your batch size. You can equate this error to CUDA out of menory. You can use devices with more GPU memories to solve the problem.