open-mmlab/OpenPCDet #1491: How to Use Argoverse2 Lidar Dataset for VoxelNeXt?

ioeddk, 2023-10-18:
Is there an official instruction for using the Argoverse2 Lidar dataset for the VoxelNeXt model? VoxelNeXe in the `tools/cfgs/argo2_models` is the only Argoverse2 model, but the instructions in GETTING_STARTED seem to use Argo2 Sensor datasets. When I download the Argo2 sensor dataset, it contains images, but the VoxelNeXt model uses point clouds isn't it?

ioeddk, 2023-11-16:
Not yet. I rather use NuScenes' model with 4 channels.

ioeddk, 2023-11-16:
> I have the same question, have you found the correct way to use the argo2 dataset for training?

Not yet. I rather use NuScenes' model with 4 channels.

yinheyanxian, 2023-11-25:
> > I have the same question, have you found the correct way to use the argo2 dataset for training?
> 
> Not yet. I rather use NuScenes' model with 4 channels.

I have implemented the training av2 data set, which just requires downloading the data set and generating the training file by following the steps in the official documentation

closed 2024-01-09 by github-actions[bot]