open-mmlab/OpenPCDet #1494: run Demo.py visulization question

thiesymon, 2023-10-26:
`(openpcdet) thiesy@ubuntu-MS-7C75:~/OpenPCDet/tools$ python demo.py --cfg_file cfgs/kitti_models/pointpillar.yaml --ckpt pointpillar_7728.pth --data_path 000008.bin
2023-10-26 13:46:21,971   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2023-10-26 13:46:21,971   INFO  Total number of samples:        1
/home/thiesy/miniconda3/envs/openpcdet/lib/python3.8/site-packages/torch/functional.py:445: UserWarning: torch.meshgrid: in an upcoming release, it will be required to pass the indexing argument. (Triggered internally at  /opt/conda/conda-bld/pytorch_1639180588308/work/aten/src/ATen/native/TensorShape.cpp:2157.)
  return _VF.meshgrid(tensors, **kwargs)  # type: ignore[attr-defined]
2023-10-26 13:46:23,601   INFO  ==> Loading parameters from checkpoint pointpillar_7728.pth to CPU
2023-10-26 13:46:23,613   INFO  ==> Done (loaded 127/127)
2023-10-26 13:46:23,626   INFO  Visualized sample index:        1
No protocol specified
[Open3D WARNING] GLFW Error: X11: Failed to open display :0.0
[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 100, in main
    V.draw_scenes(
  File "/home/thiesy/OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'`

runjiekang, 2023-10-29:
请问你解决了嘛

thiesymon, 2023-10-29:
> 请问你解决了嘛

一直没有解决啊

closed 2023-12-13 by github-actions[bot]

AJAY31797, 2024-02-11:
Hi @thiesymon, 

I am facing the same issue. Could you solve it? 

Thanks.

DongYingwei, 2024-12-26:
大家是不是都是在docker下运行出现的这个问题。

Petros626, 2025-09-13:
you're using a function which isn't available in your `Open3D` version, thats it.