open-mmlab/OpenPCDet #1495: new problem

runjiekang, 2023-10-28:
Traceback (most recent call last):
  File "demo.py", line 18, in <module>
    from pcdet.datasets import DatasetTemplate
  File "/root/OpenPCDet/pcdet/datasets/__init__.py", line 15, in <module>
    from .argo2.argo2_dataset import Argo2Dataset
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_dataset.py", line 15, in <module>
    from .argo2_utils.so3 import yaw_to_quat, quat_to_yaw
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 10, in <module>
    def quat_to_mat(quat_wxyz: Tensor) -> Tensor:
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1318, in script
    fn = torch._C._jit_script_compile(
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_recursive.py", line 841, in try_compile_fn
    return torch.jit.script(fn, _rcb=rcb)
  File "/root/miniconda3/lib/python3.8/site-packages/torch/jit/_script.py", line 1318, in script
    fn = torch._C._jit_script_compile(
RuntimeError: 
cannot statically infer the expected size of a list in this context:
  File "/root/miniconda3/lib/python3.8/site-packages/kornia/geometry/conversions.py", line 553

    # this slightly awkward construction of the output shape is to satisfy torchscript
    output_shape = [*list(quaternion.shape[:-1]), 3, 3]
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    matrix = matrix_flat.reshape(output_shape)
'quaternion_to_rotation_matrix' is being compiled since it was called from 'quat_to_mat'
  File "/root/OpenPCDet/pcdet/datasets/argo2/argo2_utils/so3.py", line 19
        (...,3,3) 3D rotation matrices.
    """
    return C.quaternion_to_rotation_matrix(
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
        quat_wxyz, order=C.QuaternionCoeffOrder.WXYZ
        ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~ <--- HERE
    )

runjiekang, 2023-10-28:
已经解决

runjiekang, 2023-10-29:
[Open3D WARNING] GLFW Error: X11: The DISPLAY environment variable is missing
[Open3D WARNING] Failed to initialize GLFW
Traceback (most recent call last):
  File "demo.py", line 112, in <module>
    main()
  File "demo.py", line 100, in main
    V.draw_scenes(
  File "/root/OpenPCDet/tools/visual_utils/open3d_vis_utils.py", line 49, in draw_scenes
    vis.get_render_option().point_size = 1.0
AttributeError: 'NoneType' object has no attribute 'point_size'
root@autodl-container-712d11903c-da40c3f3:~/OpenPCDet/tools#

runjiekang, 2023-10-29:
、

closed 2023-10-29 by runjiekang

AJAY31797, 2024-02-11:
Hi @runjiekang, 

I am facing the same problem. Could you please let me know how did you solve it?

Thanks.