open-mmlab/OpenPCDet #1498: BEVFusion eval is something awkward

ies0411, 2023-11-02:
Hello.

The results eval of my BEVFusion is lower than eval score written on pcdet reame.

_--------------average performance-------------
trans_err:       0.2848
scale_err:       0.2528
orient_err:      0.3259
vel_err:         0.2673
attr_err:        0.1860
mAP:     0.6444
NDS:     0.6905_

According to pcdet readme score, mAP should be 0.68. But My eval is quite different. 
I used default config and trained from lidar branch pretrained model.

Is there any more technic or need more epoch???

liuyuansheng624, 2023-11-20:
> Hello.
>
> The results eval of my BEVFusion is lower than eval score written on pcdet reame.
>
> _--------------average performance------------- trans_err: 0.2848 scale_err: 0.2528 orient_err: 0.3259 [...]

Hello, I have also been training bevfusion on openpcdet recently, could you please add friends together and communicate? My wechat is 13731082126

closed 2024-01-05 by github-actions[bot]

ahtchow, 2024-06-06:
same I am getting 0.6580 mAP when training BEVFusion. Anyone else getting under-performing results?

zhuoxiao-chen, 2024-10-11:
> same I am getting 0.6580 mAP when training BEVFusion. Anyone else getting under-performing results?

Hi, I used the provided pretrained Transfusion ckpt then train the BEVfusion. My batch size is 4 and trained with a single card. The results of BEVfusion are:

--------------average performance-------------
trans_err:       0.2809
scale_err:       0.2556
orient_err:      0.2986
vel_err:         0.2832
attr_err:        0.1906
mAP:     0.6466
NDS:     0.6924

Have you obtained the results which are similar to the official ones?