open-mmlab/OpenPCDet #1502: error: command ':/usr/local/cuda-11.1/bin/nvcc' failed: No such file or directory

jesanli, 2023-11-06:
when  **"python setup.py develop"**
[image]
[image]
[image]
[image]

help me to solve this question,thanks!
**I really want to use openpcdet in time.**

ammaryasirnaich, 2023-11-19:
1- First do `nvidia-smi`, you will get the detials about the nvidia driver and cuda verion if properly installed
2- Do `which nvcc` to see if the nvcc is there
3- Finally, try to export the cuda path to CUDA_HOME,  `export CUDA_HOME=/usr/local/cuda`
I hope it works

jesanli, 2023-11-20:
> 1- First do `nvidia-smi`, you will get the detials about the nvidia driver and cuda verion if properly installed 2- Do `which nvcc` to see if the nvcc is there 3- Finally, try to export the cuda path to CUDA_HOME, `export CUDA_HOME=/usr/local/cuda` I hope it works

yeah，I have solved this issue with using method 3 in terminal,Thanks

closed 2024-01-05 by github-actions[bot]