open-mmlab/OpenPCDet #1508: My custom dataset has a class that dose not exist in kitti

yux326, 2023-11-08:
I'm trying to train Pointpillar model with my own dataset whhich consists of three classes: Car, Bus, Truck.
I found in CUSTOM_DATASET_TUTORIAL.md that I have to map class names of my dataset to kitti class names, but there is no class in kitti  matches "Bus" in my dataset. How can I deal with this?

tyutlk, 2023-11-10:
just match "bus" with "car"  and it just has effects on  the evaluation.

yux326, 2023-11-10:
Thanks.
But I also have “Car" in my dataset, " there is also the class 'Car.' Will mapping 'bus' to 'car' cause confusion between these two classes?
Besides i wonder what to do while training?

closed 2023-12-26 by github-actions[bot]

Petros626, 2024-10-08:
```yaml
MAP_CLASS_TO_KITTI: {
    'Vehicle': 'Car',
    'Pedestrian': 'Pedestrian',
    'Cyclist': 'Cyclist',
}
```