open-mmlab/OpenPCDet #1509: Can't generate info for custom dataset with only one class

yux326, 2023-11-09:
My custom dataset has only one class. It returns error when runing custom_dataset.py:

> ...........
................
train sample_idx: 001101
train sample_idx: 001102
Traceback (most recent call last):
  File "/opt/conda/envs/pcdet/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/conda/envs/pcdet/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/OpenPCDet-master/pcdet/datasets/custom/custom_dataset.py", line 283, in <module>
    save_path=ROOT_DIR / 'data' / 'car_lidar',
  File "/home/OpenPCDet-master/pcdet/datasets/custom/custom_dataset.py", line 249, in create_custom_infos
    class_names, num_workers=workers, has_label=True, num_features=num_features
  File "/home/OpenPCDet-master/pcdet/datasets/custom/custom_dataset.py", line 166, in get_infos
    return list(infos)
  File "/opt/conda/envs/pcdet/lib/python3.6/concurrent/futures/_base.py", line 586, in result_iterator
    yield fs.pop().result()
  File "/opt/conda/envs/pcdet/lib/python3.6/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()
  File "/opt/conda/envs/pcdet/lib/python3.6/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/opt/conda/envs/pcdet/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/OpenPCDet-master/pcdet/datasets/custom/custom_dataset.py", line 156, in process_single_scene
    annotations['gt_boxes_lidar'] = gt_boxes_lidar[:,0:7]
IndexError: too many indices for array: array is 1-dimensional, but 2 were indexed

tyutlk, 2023-11-10:
It looks like that you have empty label file ,rather than about class nums.

yux326, 2023-11-10:
Thank you so much！
I found several empty label files,you are right!
trying to figure out how to deal with data with no targets.

closed 2023-11-10 by yux326