open-mmlab/OpenPCDet #1510: 使用官方提供的pv-rcnn模型跑demo时，只检测一半？

Elken-lab, 2023-11-10:
我发现运行demo检测只能显示x>0的部分，然后我通过修改kitti的配置文件中POINT_CLOUD_RANGE的范围，显示了完整的点云，但好像并没有对另一半进行检测？这是为什么呢？还是说就是官方提供的pv-rcnn模型检测范围就这么多？请大佬指教，谢谢谢谢谢谢谢谢谢谢谢谢！！！！！
[image]
[image]

spacewalk01, 2023-11-16:
hello, I have the same problem, How did you change the point cloud range value?

Elken-lab, 2023-11-24:
我没有找到改变检测范围的办法。我最后是通过对一个点云文件检测时检测两遍（其中一遍x取反）的方法，达到全部检测的目的的。

***@***.***
 
发件人： spacewalk01
发送时间： 2023-11-16 14:37
收件人： open-mmlab/OpenPCDet
抄送： Elken-lab; Author
主题： Re: [open-mmlab/OpenPCDet] 使用官方提供的pv-rcnn模型跑demo时，只检测一半？ (Issue #1510)
hello, I have the same problem, How did you change the point cloud range value?

SoshunNaito, 2023-12-04:
Have you checked kitti_dataset.yaml?
I was facing the same problem and found pv_rcnn.yaml contains the line `_BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml`.
In case you have modified it like `_BASE_CONFIG_: cfgs/dataset_configs/custom_dataset.yaml`, you should edit the corresponding YAML file.

751K, 2023-12-10:
把config打印出来看一下，config之间的继承会导致修改不一定成功

751K, 2023-12-10:
还有建议下次用英文描述））> 把config打印出来看一下，config之间的继承会导致修改不一定成功

closed 2024-01-24 by github-actions[bot]