open-mmlab/OpenPCDet #1514: Is dual RTX 4090 usage possible?

kbratsy, 2023-11-14:
Hello,

As part of my work, I will use OpenPCDet and similar repos in my object recognition processes. I want to use 2 RTX 4090 as Graphics Cards. However, RTX 4090 graphics cards do not have SLI support. Will this situation cause any problems in my work? So in short, is it possible to use 2 RTX 4090 graphics cards with OpenPCDet?

Have a nice day

IrumaMegumi, 2023-11-24:
Hello, you can use it at 2 GPUS. I have done it on my cloud.

rasrab1992, 2023-12-07:
Hello @IrumaMegumi.
I've tried to run this repo code using RTX 4090. But there some error when using "python setup.py develop". Did you run it without error? I think RTX 4090 needs cuda version>= 11.8 and this code work has been tested with cuda <=11.6. So could you please let me know how did you solve it?

rasrab1992, 2023-12-07:
@kbratsy 
Did you solve the issue? I also have 2 RTX 4090 graphics cards and I need OpenPCDet for my furure work as well.

kbratsy, 2023-12-08:
No, I haven't used it either, dual RTX 4090. If anyone has used it before and if this repo works, I wanted to ask because I was thinking of buying it.

IrumaMegumi, 2023-12-11:
@rasrab1992 Hello! I used my cloud server to run the code. I lowered my cuda version to run it. In my test, an epoch of PV-RCNN on 2 RTX4090 needs 5.5 minutes. However, it is 6.5 minutes on 1 RTX4090. THus, I recommend 1 RTX4090 device.

rasrab1992, 2023-12-11:
> @rasrab1992 Hello! I used my cloud server to run the code. I lowered my cuda version to run it. In my test, an epoch of PV-RCNN on 2 RTX4090 needs 5.5 minutes. However, it is 6.5 minutes on 1 RTX4090. THus, I recommend 1 RTX4090 device.

Hello @IrumaMegumi. Thank you for your response. But the issue is that RTX4090 don't work properly with CUDA <=11.8. Which version of cuda did you use?

IrumaMegumi, 2023-12-11:
[image]

closed 2024-01-25 by github-actions[bot]