open-mmlab/OpenPCDet #1518: demo.py error！

dAngelo1208, 2023-11-17:
I used demo.py to infer my own point cloud(.bin file), but account for error！
the logger display:
2023-11-17 16:17:43,642   INFO  -----------------Quick Demo of OpenPCDet-------------------------
2023-11-17 16:17:43,642   INFO  Total number of samples: 	1
2023-11-17 16:17:45,551   INFO  ==> Loading parameters from checkpoint /home/lxy/OpenPCDet-master/pv_rcnn_8369.pth to CPU
2023-11-17 16:17:45,581   INFO  ==> Done (loaded 367/367)
2023-11-17 16:17:45,715   INFO  Visualized sample index: 	1
[Exception|implicit_gemm_pair]indices=torch.Size([0, 4]),bs=1,ss=[41, 1600, 1408],algo=ConvAlgo.MaskImplicitGemm,ksize=[3, 3, 3],stride=[1, 1, 1],padding=[1, 1, 1],dilation=[1, 1, 1],subm=True,transpose=False
Traceback (most recent call last):
  File "/home/lxy/OpenPCDet-master/tools/demo.py", line 127, in <module>
    main()
  File "/home/lxy/OpenPCDet-master/tools/demo.py", line 113, in main
    pred_dicts, _ = model.forward(data_dict)
  File "/home/lxy/OpenPCDet-master/pcdet/models/detectors/pv_rcnn.py", line 11, in forward
    batch_dict = cur_module(batch_dict)
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/lxy/OpenPCDet-master/pcdet/models/backbones_3d/spconv_backbone.py", line 149, in forward
    x = self.conv_input(input_sp_tensor)
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/modules.py", line 137, in forward
    input = module(input)
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1051, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 404, in forward
    raise e
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/conv.py", line 381, in forward
    res = ops.get_indice_pairs_implicit_gemm(
  File "/home/anaconda3/envs/openpcdet/lib/python3.8/site-packages/spconv/pytorch/ops.py", line 330, in get_indice_pairs_implicit_gemm
    SpconvOps.generate_subm_conv_inds(inds_tv,
RuntimeError: /tmp/pip-build-env-zgmbfbm_/overlay/lib/python3.8/site-packages/cumm/include/tensorview/cuda/launch.h(53)
N > 0 assert faild. CUDA kernel launch blocks must be positive, but got N= 0
SPCONV_DEBUG_SAVE_PATH not found, you can specify SPCONV_DEBUG_SAVE_PATH as debug data save path to save debug data which can be attached in a issue.

michaelneri, 2023-11-22:
It is probably due to the point cloud format. is the .bin file following the KITTI dataset structure? Have you tried the demo with one sample of the KITTI dataset?

SoshunNaito, 2023-11-29:
I faced the same problem yesterday and found it was due to the range along Z-axis. In my case, my point cloud has a range of [-3.9941163, 26.552996], where most of them are located in [-3.0, 3.0]. After I removed the outliers that satisfy z < -3.0 or z > 3.0, it successfully worked.
I hope it will be helpful.

closed 2024-01-13 by github-actions[bot]