open-mmlab/OpenPCDet #1524: Error in eval pillarnet+multi_head: KeyError:  cls_preds, label_preds = torch.max(cls_preds, dim=-1) TypeError: max() received an invalid combination of arguments - got (list, dim=int), but expected one of:''

xiaomonv111, 2023-12-05:
when i eval pillarnet+multihead, there is a error
 TypeError: max() received an invalid combination of arguments - got (list, dim=int), but expected one of:''
 * (Tensor input, *, Tensor out)
 * (Tensor input, Tensor other, *, Tensor out)
 * (Tensor input, int dim, bool keepdim, *, tuple of Tensors out)
 * (Tensor input, name dim, bool keepdim, *, tuple of Tensors out)
how can i fix it
Thank you very much

xiaomonv111, 2023-12-05:
If I have already the post-processing function in the pillarnet.py file commented out
`from .detector3d_template import Detector3DTemplate

tyutlk, 2023-12-05:
just comment out the following in pillarnet.py in detectors directory when using anchor head single 

`  

    # def post_processing(self, batch_dict):
    #     post_process_cfg = self.model_cfg.POST_PROCESSING
    #     batch_size = batch_dict['batch_size']
    #     final_pred_dict = batch_dict['final_box_dicts']
    #     recall_dict = {}
    #     for index in range(batch_size):
    #         pred_boxes = final_pred_dict[index]['pred_boxes']

    #         recall_dict = self.generate_recall_record(
    #             box_preds=pred_boxes,
    #             recall_dict=recall_dict, batch_index=index, data_dict=batch_dict,
    #             thresh_list=post_process_cfg.RECALL_THRESH_LIST
    #         )

    #     return final_pred_dict, recall_dict
`

xiaomonv111, 2023-12-06:
Thank you for your reply,but I still have a question.I do not comment out the following in pillarnet.py in detectors.
when i eval pillarnet+multihead, there is a error
    final_pred_dict = batch_dict['final_box_dicts']
KeyError: 'final_box_dicts'

closed 2024-01-20 by github-actions[bot]